robotology/mesh-support #72: Flawed MOOG motors

ChiaraBartolozzi, 2015-10-07:
A set of MOOG motors, some of which have been mounted on iCubGenova02, iCubHeidelberg01, and  iCubSheffield01, have misplaced Hall effect sensors. 

The consequence of this flaw is that the position of the motor is wrongly measured and could: 
- trigger a control board failure
- lead to wrong position of the stator field, and thus a torque shortfall. This shortfall cannot be detected by the control board, i.e. no message to the user is sent in the case of a torque shortfall.

Either of these issues may occur at any time, e.g. during calibration or run-time tests. A control board failure is the worst case, which requires a power cycle and a reboot of the robot interface. -- possibly linked to issue 
A torque shortfall can compromise a specific task or eventually the robot stability. For instance, if the torque shortfall occurs during calibration, the joint may not reach the desired value within the expected time, thus leading to the calibration failure “timeout before reaching position 0...”. -- possibly linked to issue #52 (Purpley)  #37 (Sheffield) 

We need to perform tests on the Purple robot to check the effects of the flaw on the motors and characterise the behaviour in case other robots with the defective motors will encounter the same problems in the future.

We also need to fix the robot, either by changing the motors, or the rotors (with ad hoc modified rotors from MOOG) and re-run the tests to check if the problems related to the displacement of the Hall effect sensors will disappear.

For this last point we need to define which tests need to be performed (e.g. joint torque characterisation, etc.).

@hu-yue @nunoguedelha @DanielePucci @randaz81 @maggia80 @giorgiometta @Fabrizio69

hu-yue, 2015-10-07:
Hi, can I know if this issue also might affect iCubHeidelberg01 or now you're sure that this is not going to happen?

DanielePucci, 2015-10-07:
Hi @hu-yue, we are still investigating the causes of these hardware faults/torque shortfalls. So, no conclusion has been derived yet. We will keep you in the loop.

hu-yue, 2015-10-07:
Ok thanks, I'll keep an eye in case something similar happens to our robot.

closed 2018-01-16 by julijenv