robotology/mesh-support #75: HW on head joints

pattacini, 2015-10-13:
Gentlemen

We have a long-lasting issue on the head of `Bluey`: the joints go in `HW` very frequently, deeply bothering any kind of work with the robot (demos included).

Can we find a suitable time slot to seat around a table and discuss how to carry out **profitable investigations** and sort out this problem once for all?

/cc @vtikha @super-ste @randaz81 @alecive @julijenv @robotology/icub-support-admin ... and whoever may be involved

pattacini, 2015-10-14:
@super-ste we're going to attend **IFI Day 3** on 22 Oct.
Will you be willing to discuss this with us during that session?

super-ste, 2015-10-22:
Action Points from IFI:

- [ ] Get recent Logs
-  [ ] Check PWM limit in config file (we're piloting this 12V motors with 40V)
- [ ] Check Limit Current on Config File
- [ ] Check mech assy for stiction

pattacini, 2015-11-05:
Any news on this? 
Yesterday I saw again `HW` occurring on `Bluey` and on _another robot_ too.

super-ste, 2015-11-05:
Who can perform the checks stated above?
@randaz81 any Idea?
@julijenv can you and/or @spada1978 check for mechanical stiction (and maybe grease a bit) the HDs on the neck?

pattacini, 2015-11-05:
I'd give priority to the _"math"_ required for reducing the PWM limit, apply them and then perform checks.

julijenv, 2015-11-05:
It has been changed the PWM limit from 800 to 480 in the file .local/shareyarp/robots/iCubGenova04/hardware/mechanical/head-cfw2_can0-mec.xml
Let's see if it solves this sooooo long pending issue with the neck. If yes, we will add this change on the repository and github and other robots with V2 head..

pattacini, 2015-11-05:
Thanks @julijenv :+1:

julijenv, 2015-11-18:
If he head of bluey isnt making any other HW Fault we could hink of applying the change on all the other V2 head robot (especially in ours)
@super-ste , @pattacini , @randaz81 , @vtikha

pattacini, 2015-11-18:
I believe we can go ahead with the proposed change, even if it won't be decisive; it's somehow a bug fix.

Please apply the fix into the repository, not locally.

super-ste, 2016-01-04:
hi all,
has this been done? Can we close the Issue? And maybe #121 too?

pattacini, 2016-01-04:
Hi @super-ste 
Unfortunately not yet: those faults are still very annoying.

However, I think @randaz81 is preparing a new special `FW` release to patch the way the temperature control is done.

closed 2017-09-18 by pattacini