robotology/mesh-support #76: Hands issues

bhigy, 2015-10-13:
I spotted following issues with blue icub hands:
   - [ ] Left hand - ring finger: tendon is to tensed and the finger is always folded.
   - [ ] Left hand - ring finger: fingertip is not working
   - [ ] Left hand - fingers adduction/abduction : not calibrated properly and the motor is overheating
   - [ ] right hand - index: moves with saccades
   - [ ] right hand - thumb proximal: takes time before moving
   - [ ] right hand - thumb opposition: not calibrated properly (shows 20 degrees when it should be 0)
   - [ ] right hand - thumb opposition: doesn't move properly when joint position is 0 (shows as 20 because of the the calibration issue)

bhigy, 2015-10-14:
Left hand - ring finger fingertip and left elbow tendon repaired thanks to @julijenv

bhigy, 2015-10-14:
I tried to make rotations with the left forearm and the protruding tendon still touches other parts. This prevents the rotation from being performed smoothly.

bhigy, 2015-10-15:
Thumb tactile fingertip of the left hand repaired.

lornat75, 2015-10-20:
DO we still have the first issue:
tendon is too tensed and the finger is always folded

otherwise we could close this issue?

lornat75, 2015-10-20:
@julijenv

closed 2015-12-02 by bhigy