robotology/mesh-support #77: Migration from iCubInterface to RobotInterface

randaz81, 2015-10-14:
iCubLugano01 files need to upgraded from .ini format (iCubInterface) to xml format (robotInterface). Robot firmware needs to be updated too.

marijnfs, 2015-10-14:
should I be able to simply run the update script as in http://wiki.icub.org/wiki/Firmware?

marijnfs, 2015-10-15:
Hello, what are the most up-to-date documentations regarding these procedures? I can not find a clear new documentation for these new changes.

randaz81, 2015-10-16:
Sorry Marijn, upgrading from iCubInterface to RobotInterface is something which cannot be done by a icub user, since it involves the complete reorganization of the robot configuration files. We already started the upgrade process: the old iCubLugano01 folder has been deleted, and the new template has been committed. The new xml files will be written by hand, so it will take few days of additional work.

marijnfs, 2015-10-18:
Ok I see, thank you for the effort! I'll wait eagerly for the files;)

regards,
Marijn

randaz81, 2015-10-19:
calibration files pushed to the repo

randaz81, 2015-10-21:
New configuration files ok (tested remotely)
There's an issue on joint 10 of the left arm (thumb distal), the joint does not come back. @robotology/icub-support-admin @julijenv can you give a look?

julijenv, 2015-10-29:
Hi @marijnfs ,
please let come on skypewith a webcam and we will try to understand the little issue you have with the joint 10 of the left arm:
our skype ID : icub.support

closed 2018-01-16 by julijenv