robotology/mesh-support #78: [iCubDarmstadt] Robot setup in view of CoDyCo Year 3 Review meeting

francesco-romano, 2015-10-14:
This will contain all the issues (HW and SW) on iCubDarmstadt.

Please @robertocalandra (and Alex and Elmar) can you please start filling this issue with the problems iCub has at the current state?

cc @jeljaik @DanielePucci

DanielePucci, 2015-10-14:
As discussed with Jan, @robertocalandra  and  @AlexandrosParaschos should accomplish  the following tasks before @francesco-romano and @jeljaik leave for Darmstadt:

- [ ] Update the software of: firmware, yarp, icub-main, and codyco-superbuild; the update should be done one week before our visit;
- [ ] Do quantitative tests for kinematics measuring devices: check that all encoders, both on motor and joint side, work properly.  Also, check that the IMU respond reasonably;
- [ ] Do qualitative tests for dynamics measuring devices: check that all  force/torque sensors work reasonably;

lornat75, 2015-10-20:
Please open individual issues for each problem, itemized multi-issues are difficult to track. Group issues with labels (we can create additional labels) to simplify searching.

DanielePucci, 2015-10-21:
Ack. However, the above items are not issues, but tasks. So, I think we can keep this issue open to track tasks, and open individual issues for eventual HW or SW problems.

robertocalandra, 2015-10-22:
Ok then we will just post links to the issues.

Please note: #25 still open

marcoaccame, 2015-10-22:
About the software update ... 
@robertocalandra can you please also verify if you have the latest kernel w/ RT features?
I believe you already have because together with Rueckert Elmar in July we performed an update of the SW which works with RT kernel, but ...

you just need to digit (as explained in #87):

```
less /VERSION_PC104
```

rueckert, 2015-10-27:
Hi, 

@marcoaccame our version is 5.0-15.03.31

marcoaccame, 2015-10-28:
Ok, thanks.

you should update the live key to latest version, which is now 5.0-15.10.26 following instructions in http://wiki.icub.org/wiki/The_Linux_on_the_pc104

DanielePucci, 2015-11-06:
Two weeks to go, tasks 2 and 3 shall be considered of highest importance.

robertocalandra, 2015-11-11:
Kernel updated to 5.0-15.11.02. However, after updating also the firmware and the repositories, the robotInterface fail during initialization as detailed in #108

DanielePucci, 2015-11-14:
What about the tasks 2 and 3?

paraschos, 2015-11-16:
The tasks 2 and 3 were completed as accurately as their description.

closed 2015-11-16 by robertocalandra