robotology/mesh-support #79: Joint 4 of the Right Leg "broke" after today's demo

jeljaik, 2015-10-14:
`robotInterface` shows an error regarding joint 4 of the right leg. Also, although that's the only joint in hardware fault state, the yaw joint of the leg (joint 2 I believe) starts rotating inwards during the calibration  procedure. 

@randaz81 As we were discussing earlier, I just wanted to put my two centsþ on the `icub-support` repository. First, I think you might wanna add a `CONTRIBUTING.md` file in the root of this repo where you explain the right way to label the issues or where some standard way to describe the problem is detailed. When this is done, everytime someone files an issue, the following note: **Please review the guidelines for contributing to this repository** appears on top as in this image.

[image]

Something else that can be done for better compilation of the issues is a script like the one I included in the `WBI-Toolbox` repo to have some sort of template issue with either precompiled information, like environmental variables of the user's system (but this might be more desirable for a purely software repo) or simply other fields to be compiled by the user with some *default* information on the hardware. 

https://github.com/robotology-playground/WBI-Toolbox/blob/master/create_issue.sh

pattacini, 2015-10-14:
@jeljaik the `CONTRIBUTING.md` file is a very good idea indeed, thanks! We should adopt it for a number of repositories of our eco-system.

Conversely, I deem the script a bit too much for `icub-support`.

jeljaik, 2015-10-14:
:+1: Sure thing. Maybe the script would be best suited for `icub-main` or more software related repos.

pattacini, 2015-10-14:
Use of `CONTRIBUTING.md` endorsed for [**`icub-support`**](https://github.com/robotology/icub-support) and [**`QA`**](https://github.com/robotology/QA).

Thanks @jeljaik

lornat75, 2015-10-20:
Can you explain to me the difference between the contribution.md and readme.md?

francesco-romano, 2015-10-20:
If you have a Contribution.md the link to it will appear when you open an issue:

[image]

pattacini, 2015-10-20:
It's a somewhat undocumented feature indeed, quite useful though.

jeljaik, 2015-10-22:
This issue was fixed by @julijenv. Closing.

closed 2015-10-22 by jeljaik