robotology/mesh-support #81: eyelids joint

valegagge, 2015-10-14:
if I move the joint with high velocity , it seems that the two weels slide.

valegagge, 2015-11-13:
Marcello worked on this issue and reported:
"Azione: La vite che serrava il motore non era completamente stretta, si è riposizionato il motore e si è serrato il supporto che lo tiene in posizione.
E’ stata inoltre riscontrata una anomalia nel pezzo che tiene il sensore . il supporto a codice RC_IIT_023_P_027 in questione è stato sostituito, in quanto quello montato era storto e il magnete non era parallelo al sensore.
Risultato: Da testare se si è risolto il problema."

The problem is not solved completely.
The screew too tight creates a lot of frinction and not allowing orderly movement of the joint.
In orter to reduce friction I asked to production guys if it is possible to file a small step on plastic support.

maggia80, 2015-11-25:
If the magnet is too far from the AEA sensor (out of the specs), the AEA gives strange value that are recognized by the firmware as "SPIKE" , because there were two consecutive reading with a difference higher than 0.615 degree in one ms (https://github.com/robotology/icub-firmware/issues/18).  I suggest to change the AEA support material from plastic to metal.

super-ste, 2015-12-22:
we'll beef it up a little and change the material to FDM ABS instead of resin; that sould do.

super-ste, 2015-12-28:
New beefen-up has been designed-built and assembled.
@valegagge cna you test the eyelids movement and let us know if the issue is solved?

valegagge, 2016-01-11:
The issue is not solved: fw continues to recognize spikes, even if @maggia80 moved motor near to sensor.
Maybe the sensor is not centered perferctly or the sensor is damaged.

valegagge, 2016-01-21:
I srcewed little better the screw on the motor and now there aren't more spikes.

closed 2016-01-21 by valegagge