robotology/mesh-support #83: Errors after updating sw and fw on iCubHeidelberg01

hu-yue, 2015-10-14:
Hi, I updated all software and firmware for iCubHeidelberg01 and get tons of errors and warnings when launching robotInterface so the robot won't start.
I think there are some parameters missing in the configuration files, however, I don't know which ones are missing from where and if it's just this the problem.
Please find the log here: https://www.dropbox.com/sh/ovzcx28jawt19m7/AACKF2xRFShFQaSvTuIhAn7ka?dl=0

cc @randaz81

hu-yue, 2015-10-14:
By cross-checking with iCubGenova02 configuration files I think I've filled in the missing parameters (copy-pasted from Genova02), now the errors regarding missing parameters are gone but the robot cannot start anyways due to other errors and gets stuck on warnings.
The new log is always here: https://www.dropbox.com/sh/ovzcx28jawt19m7/AACKF2xRFShFQaSvTuIhAn7ka?dl=0

randaz81, 2015-10-14:
Hi Yue, looking at the log, I noticed that the error code is "wrong protocol". Therefore my first guess is that not all the boards have been updated with the correct firmware. e.g. you may have updated only the ems boards and not the 2foc (or the skin) boards. Since the EMS protocol changed, if the 2foc boards have not been updated too, they will be not able to communicate anymore.

hu-yue, 2015-10-15:
Hi,
Yes yesterday I did not update the CAN fw, today I updated everything (2FOC, strain and skin) and I get calibration errors. The new log is always in here: https://www.dropbox.com/sh/ovzcx28jawt19m7/AACKF2xRFShFQaSvTuIhAn7ka?dl=0

hu-yue, 2015-10-15:
Hi,
I updated also the CAN fw (2FOC, strain and skin), the robotInterface can be launched and the calibration is done, but some errors (and warnings) appear in the log.
The new log is here: https://www.dropbox.com/sh/ovzcx28jawt19m7/AACKF2xRFShFQaSvTuIhAn7ka?dl=0

hu-yue, 2015-10-15:
It seems that the remaining error is about packet loss, it happens from time to time that the error
```
hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.5

```
appears for different boards.

cc @randaz81 @marcoaccame

lornat75, 2015-10-20:
We should check if the socket buffer size needs to be tuned or install the RT linux.

hu-yue, 2015-10-20:
If I'm not wrong RT linux has been installed when the robot was brought back to IIT in May 2015.

hu-yue, 2015-10-20:
I just checked, this is what is on the pc104 now: 3.2.0-4+icub.1-rt-amd64

marcoaccame, 2015-10-22:
Hi Yue, there should be version 5.0-15.04.14
see issue #87 about how to see the correct version.

hu-yue, 2015-10-22:
Just checked and replied on issue #87, any comment/progress about the packet loss error?

marcoaccame, 2015-10-22:
what is the value of ICUB_REALTIME_EXPERIMENTAL in your build?
it should be set to 1 to give correct priorities to the tx and rx threads

marcoaccame, 2015-10-22:
asked @apaikan help so that you can run a script which detects OS receive buffer size and other net configurations of the system. see issue #8 to read what it is about.

hu-yue, 2015-10-22:
@marcoaccame ICUB_REALTIME_EXPERIMENTAL was ON on pc104 and OFF on the cluster.
I recompiled on the cluster with ON and for the time being I have not seen the error about packet loss.

hu-yue, 2015-10-22:
Ok, the error just re-appeared for boards 6,7,8 and 9, at the same time. 
Log if needed: https://www.dropbox.com/sh/ovzcx28jawt19m7/AACKF2xRFShFQaSvTuIhAn7ka?dl=0
@marcoaccame

marcoaccame, 2015-10-22:
It is correct to be ON only on pc104 where robotInterface runs

hu-yue, 2015-10-23:
Ok. But the problem remains either way. @marcoaccame

marcoaccame, 2015-10-26:
Hi Yue, I looked at the log files. 

I have seen a regular pattern: the runs of sequence number errors are separated by either 0 ms, 10 ms or 20 ms. And they always give 1 UDP frame missing. 

I have suspects, but I need more evidence. Would you please do the following?

- use top to see the % of CPU of processes robotInterface and irq/20-eth0. And please report to me (a screenshot is fine).
- if you do regular requests to embObjMotionControl every 10 ms or less, can you try to decrease them for instance at 20 ms? In this case, see % of CPU and if error happens please send me the log.
- in file hardware\electronics\pc104.xml, set PC104RXrate to have value 2 (now is 5) and see how things go. also see % of CPU in this case. if errors in sequence number happens, please send me the log.

In the meantime, I will talk to @apaikan about the network script.
 
Thank you.

hu-yue, 2015-10-26:
@marcoaccame a naive question maybe, but how are requests sent to embObjMotionControl?
In the last log, the errors appeared while the robot was doing nothing (robotInterface was running but we were not performing any task or similar).
I will provide the logs by tomorrow.

hu-yue, 2015-10-27:
@marcoaccame 

- Screenshot for the first point
[image]

- For the second point I tried to send commands every 5 ms and no errors occured, only warnings, I increased the time and no errors occured, so I'm not sure about how to tackle this point

- For the third point, this is the screenshot with PC104RXrate = 2, also in this case errors appeared for no apparent reason (we were not performing any action), log is here: https://www.dropbox.com/s/ie9kgfy4hawn534/RXRate2_ypmrprunlog_27_10_2015_03_11_08.log?dl=0, screenshot was taken before the error appeared
[image]

I still think errors appear randomly, since I cannot understand when they appear, sometimes they appear soon after I switch on the robot, sometimes after a while, sometimes without doing anything and sometimes after/during some actions.

apaikan, 2015-11-02:
@hu-yue  Would you please do the following checks: 

* open a terminal on PC104 and check that the return value of the following command is **8388608**! 
```
$ cat /proc/sys/net/core/rmem_max 
```
* check the `CMAKE_BUILD_TYPE =` **Release** for the Cmake in the Yarp and iCub build directory!

hu-yue, 2015-11-03:
@KevinAStein could you try to perform the above tasks and post the answer here? Thanks! (otherwise I will do that tomorrow)

KevinAStein, 2015-11-03:
ok, I will have a look on the issue and check what I can do after lunch

KevinAStein, 2015-11-03:
I checked and the return value is 8388608.  Both are build with ```CMAKE_BUILD_TYPE =``` Release

hu-yue, 2015-11-16:
Hi @marcoaccame @apaikan are there any further suggestions about this issue?

marcoaccame, 2015-11-20:
Hi @hu-yue, I am working on adding extra info when there is an error in sequence number. 

I will come back to you next week after I have tested the new feature and pushed it into master.

marcoaccame, 2015-11-23:
Hi @hu-yue, I have just released new SW/FW which prints some more info when robotInterface detects an error in received frames. It now also tells if one UDP frame was discarded because not fully formed. 

You need to update icub-main, icub-firmware-shared, and icub-firmware-build. Then recompile on PC104 and load new FW (version 2.19) on the EMS boards.

Please give it a try and then send me the log.

hu-yue, 2015-11-23:
Hi @marcoaccame thank you for the update.
I just want to make sure one thing: is it safe now to update the firmware? During the last update, I had to wait because the first release was somehow not compatible with my robot and I had to wait for it to be fixed. And I've been recently told to wait to update it, so should I actually proceed with the update?
Thanks

@randaz81

hu-yue, 2015-11-23:
Hi @marcoaccame I did all the updates, [here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/after_update_ypmrprunlog_23_11_2015_07_11_53.log) is the new log with some errors appeared.

marcoaccame, 2015-12-11:
Hi @hu-yue, I have released a new experimental FW version for EMS (icub-firmware-build/testing/ems-ethmon-v2.22.hex) which every 100 ms checks the status of the ETH ports and reports about possible errors. So far, we check only for link up/down and RX CRC errors.

I suspect that the lost packet you seldom see may be due to some noise in the ETH bus and the check that I have implemented helps detecting this sort of problems.

I have tested it and it works on our iCubGenova02 robot which is a superset of yours. However, I did not have much availability of robots to test it long enough to promote it to an official FW version. I will do it next week and I will tell you when ready.

But if you want to beta test the ems-ethmon-v2.22.hex straight away you are welcome. You also need updating icub-firmware to take full advantage of new diagnostics.

marcoaccame, 2015-12-11:
Hi @hu-yue, I have tested and released a new version 2.22 application. Please see if there are new diagnostics messages around the time of the packet loss.

hu-yue, 2015-12-14:
Hi @marcoaccame  thanks for the update, I updated the firmware today.
However before the usual packet loss error appeared, I got an error on torso overcurrent error, for which I will open a new issue.

But after I switched off and on again the overcurrent error did not appear, but the packet loss did, [here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/logs_issue_83/ypmrprunlog_14_12_2015_03_54_51.log) is the log.

marcoaccame, 2015-12-21:
Hi @hu-yue, I have added another check in the EMS application. There is now version 2.23. You should update the EMS boards w/ this new fw and also pull the latest version of icub-firmware-shared (and recompile robotInterface).

Then, please send me the log.

The new check in the FW verifies if every UDP packet which is written in the ETH registers (and hence the lowest level of the FW before the ETH frame gets inside the network) has a correct ever increasing sequence number as imposed by the application. 

If this test is positive and diagnostics i printed, then we must look for inside the FW. Else, for sure the packet gets lost inside the ETH network or inside the PC104 and we shall add proper additional diagnostics to find out where the packet gets lost.

hu-yue, 2015-12-21:
Hi @marcoaccame thanks for the update.
[Here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/logs_issue_83/ypmrprunlog_21_12_2015_02_49_22.log) is a log with errors after the update. It's a log with lots of activities (many warnings), if you need a "cleaner" log let me know.

I noticed that after this update the position control (not position direct) works differently.
Before, if I used a time step of 10ms there were warnings, but the motion was smooth. Now the motion is not smooth at all (jerky, "a scatti").
Is this the correct behaviour of position control?

marcoaccame, 2015-12-21:
Hi @hu-yue, I will have a look at the log asap. 

About your problems w/ position control @ 10 ms: i doubt that the changes I did on ETH diagnostics have triggered what you see. I will however look at other changes maybe done by other developers in between versions 2.22 and 2.23.

However, position control @ 10 ms is not to be used, thus we cannot guarantee a smooth trajectory.  

I let @randaz81 say something about that.

hu-yue, 2015-12-21:
Thanks @marcoaccame , yes I also figured out it was not a firmware problem. 
I know it should not be used with 10 ms, but this curious difference in behaviour is just a bit weird for me.

marcoaccame, 2015-12-21:
Hi @hu-yue, I have seen from your log a line with:
```
25.564063
5
<#STRING_START#> from BOARD 10.0.1.9 (rightFoot), src LOCAL, adr 0, time 35s 73m 676u:
 (code 0x06000003, par16 0x0001 par64 0x0000000000000179) -> 
eoerror_INTERNALERROR__valueisempty + .
<#STRING_END#>
```

Did you perform on icub-firmware-shared a git pull and then recompiled icub-main? 
Otherwise I cannot explain why an error code 0x06000003 is not parsed correctly.

hu-yue, 2015-12-21:
@marcoaccame I performed a pull on icub-firmware-shared but I think I forgot to recompile icub-main.
Sorry, I will do this again and send you a new log.

hu-yue, 2015-12-21:
@marcoaccame [Here](http://orb.iwr.uni-heidelberg.de/ftp/icub_logs/logs_issue_83/ypmrprunlog_21_12_2015_06_15_38.log) is the new log, no activities have been performed (except for launching wholeBodyDynamicsTree).

hu-yue, 2016-05-31:
Hi, it has been a while and in the meantime I have updated the robot some times, but we keep having this error on `hostTransceiver()`. 
In particular, I've notice today that when the `yarprobotinterface` just started, a series of these errors appeared in sequence:

```
870		16.472544		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.11. Expected: 201, Received: 202, Missing: 1, Prev Frame TX at 23600638 us, This Frame TX at 23606638 us 
897		16.714278		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.9. Expected: 795, Received: 796, Missing: 1, Prev Frame TX at 23841643 us, This Frame TX at 23851643 us 
898		16.714317		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.11. Expected: 284, Received: 286, Missing: 2, Prev Frame TX at 23849638 us, This Frame TX at 23858638 us 
901		16.724388		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.10. Expected: 488, Received: 489, Missing: 1, Prev Frame TX at 23854638 us, This Frame TX at 23860638 us 
902		16.724423		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.6. Expected: 1460, Received: 1461, Missing: 1, Prev Frame TX at 23853646 us, This Frame TX at 23863648 us 
905		16.734349		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.11. Expected: 289, Received: 290, Missing: 1, Prev Frame TX at 23864638 us, This Frame TX at 23870638 us 
906		16.734417		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.10. Expected: 493, Received: 494, Missing: 1, Prev Frame TX at 23869638 us, This Frame TX at 23875638 us 
909		16.744486		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.10. Expected: 495, Received: 496, Missing: 1, Prev Frame TX at 23875638 us, This Frame TX at 23881643 us 
911		16.764558		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.8. Expected: 1240, Received: 1241, Missing: 1, Prev Frame TX at 23871647 us, This Frame TX at 23881646 us 
913		16.764667		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.10. Expected: 503, Received: 504, Missing: 1, Prev Frame TX at 23899638 us, This Frame TX at 23905638 us 
915		16.774698		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.6. Expected: 1471, Received: 1472, Missing: 1, Prev Frame TX at 23908646 us, This Frame TX at 23918648 us 
922		16.986088		ERROR		hostTransceiver()::onMsgReception() detected an ERROR in sequence number from IP = 10.0.1.6. Expected: 1512, Received: 1513, Missing: 1, Prev Frame TX at 24113648 us, This Frame TX at 24123646 us 
```
I have restarted `yarprobotinterface` and they appeared again in the same way. They don't seem to have an effect on our current use of the robot, however it would be nice to know what is causing this and if we can solve it.
Thanks!

hu-yue, 2018-01-10:
I think this is not occuring anymore, mostly now after #504.

closed 2018-01-10 by hu-yue