robotology/mesh-support #84: Linking issu: "/usr/bin/ld: cannot find -lportaudio"

johnnass, 2015-10-14:
Hello, 
I am working on iCub head and want to compile an example code similar to one provided in the tutorial (tutorial_arm.cpp). 
Then I meet the flowing error in linking... 

icub@icubsrv:~/icub-tutorials/motor-control-move-head$ make 
[100%] Building CXX object CMakeFiles/move_head.dir/move_head.cpp.o
Linking CXX executable move_head
/usr/bin/ld: cannot find -lportaudio
collect2: ld returned 1 exit status
make[2]: *** [move_head] Error 1
make[1]: *** [CMakeFiles/move_head.dir/all] Error 2
make: *** [all] Error 2

Can you please tell me how to find the library  “lportaudio” or solve the problem? 

Thank you :)

pattacini, 2015-10-14:
Hi @johnnass 
I've marked this issue as **invalid** since we try to keep track here of maintenance requests for our robots (see our [**guidelines for contributing**](https://github.com/robotology/icub-support/blob/master/CONTRIBUTING.md) to this repository).

Would you mind to open an equivalent ticket into [**`icub-tutorials/issues`**](https://github.com/robotology/icub-tutorials/issues) system instead?

johnnass, 2015-10-14:
Hi 
Thank you, I will open an equivalent ticket into  icub-tutorials/issues.

pattacini, 2015-10-14:
Thanks @johnnass 
Closing in favor of https://github.com/robotology/icub-tutorials/issues/1.

closed 2015-10-14 by pattacini