robotology/mesh-support #86: Right knee never goes to 0

hu-yue, 2015-10-20:
Hi,
I've recently noticed that the right knee joint never goes to 0, e.g. the encoder always reads -1.3, if I try to drag the joint to 0 with the motor gui (which home position is set to 0 for knee joint) it jumps back to -1.3.
It is not a gui problem, in the sense that I see about -1.3 also from ``` /icub/right_leg/state:o ```.
What might be the reason? What can I check to solve the problem?
Thanks!

randaz81, 2015-10-20:
if you put the joint in idle, and you manually push it against the hardware limit, what is the encoder reading?

hu-yue, 2015-10-20:
The same, I pushed it to the mechanical limit and it's still -1.3

hu-yue, 2015-10-20:
Does this mean that the offset in the calibrationDelta is too big or something like that?

hu-yue, 2015-10-20:
I changed the value in the calibrationDelta and now it's going to 0.

closed 2015-10-20 by hu-yue