robotology/mesh-support #87: Make sure all ETH robots have the the latest icub-live v5

marcoaccame, 2015-10-22:
See issue #8

marcoaccame, 2015-10-22:
The version is written in file /VERSION_PC104 (as from http://wiki.icub.org/wiki/The_Linux_on_the_pc104).

Thus on a terminal one should digit

```
less /VERSION_PC104
```

and read version

```
5.0-15.04.14
```

I have verified that the version is correct (5.0-15.04.14) on iCubGenova02 and iCubGenova04. 

It is to be tested on iCubHeidelberg01, iCubDarmstadt01, iCubSheffield01

marcoaccame, 2015-10-22:
I am quite sure that both iCubDarmstadt01 and iCubSheffield01 already have the RT icub-live version because of some related redmine issues.

However, I shall ask for verification.

hu-yue, 2015-10-22:
Just verified on iCubHeidelberg01, the version is 
```
5.0-15.04.14
```

marcoaccame, 2015-10-22:
@urielmtz tells me that version is  5.0-15.03.31. It has RT features, but latest version is 5.0-15.04.14, thus we may update it.

marcoaccame, 2015-10-28:
@rueckert in issue #78 wrote that the iCubDarmstadt01 robot has the 5.0-15.03.31 version. Can he (or someone else in Darmstadt) update it to the latest version (which today is 5.0-15.10.26) and let me know when done? Thanks.

marcoaccame, 2015-10-30:
@urielmtz in Sheffield is doing the upgrade to latest icub-live version. see #96.

mbrunettini, 2015-11-04:
Hi @urielmtz, how's going with the iCub PC104 upgrade ( @marcoaccame  )

urielmtz, 2015-11-04:
Hi @mbrunettini ,

We have the pc104 upgraded to version 5.0-15-10-26. The problem we have is when we run icub-cluster.py icubsrv is ticked but pc104 isn't and then robotInterface doesn't start.

Cheers
Uriel

marcoaccame, 2015-11-04:
hi @urielmtz, can you please open it on another issue? in this way we keep the problems separated and there will be someone more skilled than us on icub-cluster ;-) who will be able to help you.

urielmtz, 2015-11-04:
Hi @marcoaccame, good I'll open another issue then.
Thanks.

Cheers
Uriel

marcoaccame, 2015-11-16:
As @robertocalandra in issue #78 writes ("Kernel updated to 5.0-15.11.02.") also iCubDarmstadt01 has latest live.

closed 2015-11-16 by marcoaccame