robotology/mesh-support #88: ethResources verifyBoardPresence cannot reach BOARD 3

urielmtz, 2015-10-22:
Hi all,

We updated the firmware in our iCub and we have and eth resources issue. The version of PC104 is 5.0-15.03.31. Please see log file.

||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/robotInterface.ini] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/./icub_all.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml
yarp: Port /log/pc104/robotInterface/3185 active at tcp://10.0.0.2:10005
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./cartesian/left_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./cartesian/right_arm-cartesian.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/head-cfw2_can0-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/motorControl/head-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/torso-ems5-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/torso-ems5-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/torso-ems5-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/motorControl/torso-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/left_upper_arm-ems1-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/left_upper_arm-ems1-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/left_lower_arm-ems2-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/left_lower_arm-ems2-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/left_lower_arm-ems2-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/motorControl/left_arm-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/right_upper_arm-ems3-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/right_upper_arm-ems3-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/right_upper_arm-ems3-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/right_lower_arm-ems4-mc.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/right_lower_arm-ems4-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/right_lower_arm-ems4-mec.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/motorControl/right_arm-mc_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/FT/left_arm-FT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/FT/right_arm-FT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/FT/left_upper_arm-ems1-strain.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/FT/right_upper_arm-ems3-strain.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/right_upper_arm-ems3-eln.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/VFT/left_arm-VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/VFT/right_arm-VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/VFT/torso-VFT_wrapper.xml 
[DEBUG]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/MAIS/left_hand-mais_wrapper.xml 
[WARNING]Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/MAIS/left_hand-mais_wrapper.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
[WARNING]Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/MAIS/left_hand-mais_wrapper.xml at line 4 . "devices" element should contain the "robot" attribute 
[... 339 lines cut ...]
[... cut ...]
[WARNING]FIXME: abort action not yet implemented
[WARNING]FIXME: abort action not yet implemented
[WARNING]FIXME: abort action not yet implemented
[WARNING]FIXME: abort action not yet implemented
[INFO]All actions for action level 1 of interrupt3 phase started. Waiting for unfinished actions. 
[INFO]All actions for action level 1 of interrupt3 phase finished. 
[INFO]interrupt3 phase finished. 
[WARNING]ethResources::verifyBoardPresence() cannot reach BOARD 3 at attempt # 65 w/ timeout of 0.5 seconds 

Cheers
Uriel

marcoaccame, 2015-10-22:
At first there are some errors on cfw2 which prevent mc device on can to work.

```
[ERROR]checkFirmwareVersions() failed. CanBusMotionControl::open returning false, 
``` 

Then there are some errors due to missing parameters in config files, which cause the mc devices of torso and arms to fail.

See:
```
[ERROR]RotorMax parameter not found 
[ERROR]Missing parameters in config file 
yarpdev: ERROR driver was found but could not open
[WARNING]Cannot open device torso_mc 
[WARNING]Cannot open device torso_mc 
```

```
[ERROR]RotorMax parameter not found 
[ERROR]Missing parameters in config file 
yarpdev: ERROR driver was found but could not open
[WARNING]Cannot open device left_upper_arm_mc 
[WARNING]Cannot open device left_upper_arm_mc 
```

```
[ERROR]RotorMax parameter not found 
[ERROR]Missing parameters in config file 
yarpdev: ERROR driver was found but could not open
[WARNING]Cannot open device right_lower_arm_mc 
[WARNING]Cannot open device right_lower_arm_mc 
```

After that:
```
[ERROR]
/---------------------------------------------------\ 
| Unable to bind to local IP address 18.152.0.0:12345 
---------------------------------------------------/ 
[ERROR]embObjAnalogSensor::open() fails because could not instantiate the ethResource board 1 ... unable to continue 
yarpdev: ERROR driver was found but could not open
[WARNING]Cannot open device left_upper_arm_strain 
[WARNING]Cannot open device left_upper_arm_strain 
```

It can be something related to missing params. Maybe @randaz81  can help.

urielmtz, 2015-11-04:
Hello,

We updated pc104 and also the firmware, and we have different errors (see below) when we run the robotInterface.

||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/robotInterface.ini] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/robotInterface.ini
||| finding file [config]
||| checking [/home/icub/./icub_all.xml] (pwd)
||| checking [/home/icub/.local/share/yarp/robots/iCubSheffield01/./icub_all.xml] (robot)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./icub_all.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./cartesian/left_arm-cartesian.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./cartesian/right_arm-cartesian.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/head-cfw2_can0-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/motorControl/head-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/torso-ems5-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/torso-ems5-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/torso-ems5-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/motorControl/torso-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/left_upper_arm-ems1-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/left_upper_arm-ems1-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/left_lower_arm-ems2-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/left_lower_arm-ems2-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/mechanicals/left_lower_arm-ems2-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./wrappers/motorControl/left_arm-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/motorControl/right_upper_arm-ems3-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/right_upper_arm-ems3-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubSheffield01/./hardware/electronics/pc104.xml 
[... cut ...]
[[01;33mWARNING[00m]FIXME: abort action not yet implemented
[[01;33mWARNING[00m]FIXME: abort action not yet implemented
[[01;34mINFO[00m]All actions for action level 1 of interrupt3 phase started. Waiting for unfinished actions. 
[[01;34mINFO[00m]All actions for action level 1 of interrupt3 phase finished. 
[[01;34mINFO[00m]interrupt3 phase finished. 
[[01;33mWARNING[00m]ethResources::verifyBoardPresence() cannot reach BOARD 4 at attempt # 41 w/ timeout of 0.5 seconds 
Aborting (calling abort())...

marcoaccame, 2015-11-04:
There is the same error as I wrote in my previous comment: param RotorMax is missing.

marcoaccame, 2015-11-05:
Dear @urielmtz, the parameter RotorMax is a param introduced in icub-main version 1.2.17.

This param is missing in some files, thus robotInterface does not complete. That is the reason of the failure. All other ERROR prints are caused by that.

I would rather close this issue and continue discussion in issue #104.

closed 2015-11-05 by marcoaccame