robotology/mesh-support #89: Impossibility of increasing torque saturation threshold above 24

hu-yue, 2015-10-22:
I tried to increase the torque saturation limit by changing the TorqueMax in ```hardware/mechanicals/part*.xml``` with the following values:
- 50, the saturation warning was again on 24. In this case, the warning appeared several times before the joint went to hw fault.
- 24 (default that was in the file), saturation on 24. In this case, the warning appeared and the joint went immediately on hw fault.
- 20, saturation on 20. As the first case, also in this case the warning appeared several times before hw fault.

The exact warning we get is:
```
embObjMotionControl::updateMeasure() torque measure saturated to +20.0000 on board 6 joint 1, count: 534

```
Seems that the joint is left hip roll.

hu-yue, 2015-10-22:
Ok, I think this issue can be close, because now that I can see on which board the saturation is happening and changed the TorqueMax for all the parts, it is taking the new value.
The problem before was that I could not see and due to issue #90 I thought it was always on the left leg.

closed 2015-10-22 by hu-yue

DanielePucci, 2015-10-22:
Thanks @hu-yue! The hypothesis made by @barbalberto was indeed right :+1: