robotology/mesh-support #90: Weird HW fault

hu-yue, 2015-10-22:
Hi, there has been a hw fault with the following error:
```
from BOARD 6, src LOCAL, adr 0, time 284s 770m 577u: (code 0x0200000c, par16 0x0001 par64 0x0000000100020000) -> MC: generic motor error (see 64 bit mask parameter) + INFO = NO MORE
```
```
Unable to set the control mode of board 6 joint 1 in HW_FAULT
```
The error occurred during single foot torque balancing on the left leg.
The joint in hw fault is left hip roll.

marcoaccame, 2015-10-22:
@ale-git : can you comment?

DanielePucci, 2015-10-22:
@hu-yue, did you see the problem also on two-feet balancing?

hu-yue, 2015-10-22:
I forgot this other error, which comes also before the hw fault:

```
from BOARD 6, src CAN1, adr 2, time 286s 512m 505u: (code 0x02000002, par16 0x0001 par64 0x0000000100020000) -> MC: 2FOC i2t limit exceeded + INFO = NO MORE
```

hu-yue, 2015-10-22:
@DanielePucci no I did not try, I can try now

DanielePucci, 2015-10-22:
Yes, thanks :) @hu-yue

hu-yue, 2015-10-22:
On two feet it did not happen. I tried my best to push he robot in all directions and the only thing happened is that the torso roll had torque saturation.

hu-yue, 2015-10-23:
I just tried the balancing on right leg, and also the right hip roll went in hw fault.
These the errors:
```
from BOARD 8, src CAN1, adr 2, time 442s 171m 507u: (code 0x02000002, par16 0x0001 par64 0x0000000100020000) -> MC: 2FOC i2t limit exceeded + INFO = NO MORE

from BOARD 8, src LOCAL, adr 0, time 442s 171m 586u: (code 0x0200000c, par16 0x0001 par64 0x0000000100020000) -> MC: generic motor error (see 64 bit mask parameter) + INFO = NO MORE
```
@DanielePucci

DanielePucci, 2015-10-23:
Well, waiting for @marcoaccame @ale-git opinions.

DanielePucci, 2015-10-23:
@randaz81

ale-git, 2015-10-23:
Maybe the I2T limit is too conservative for this task. I've to discuss a little bit with @randaz81 about this issue, then we'll probably able to increase the I2T limit.

DanielePucci, 2015-10-23:
So, @ale-git tell us when you are done, so @hu-yue can update the software and check.

hu-yue, 2015-11-16:
Hi @ale-git any update about this issue?

ale-git, 2015-11-16:
@hu-yue we're working on a more precise management of motor current limits in the next releases of the firmware... in the meanwhile, plese try to increase the Currents parameter in this section of 
hardware/motorControl/left_upper_leg-ems6-mc.xml
hardware/motorControl/right_upper_leg-ems8-mc.xml
Please make also sure to increase it by small steps (about 500) and don't grow too much (stay below 8000). I hope this momentary workaround will fix the problem.

    <group name="LIMITS">
        <param name="Max">              85            70            70          0       </param>
        <param name="Min">             -30           -15           -70         -100     </param>
        <param name="Currents">         5000          5000          5000        5000    </param>
    </group>

hu-yue, 2015-11-16:
@ale-git thanks, for the moment I changed the limit of the ```hip_roll``` joint to ```6500``` and the robot is not going into hw fault anymore.
I will update the firmware for the next release then and keep these changes in my local files.

closed 2018-01-16 by julijenv