robotology/mesh-support #91: Torso fault

kt10aan, 2015-10-23:
Hi,

We are having troubles with the torso of our iCub. Everything started two weeks ago, when the robot was operational (but not moving much) and then suddenly it kind of "fainted". I.e. It fell forward (across its pitch axis) but not so abruptly like when the red button is pressed, a bit slower.
During reboot, the CAN ID was changed. Fixing that fixed the issue. A week after that, the same thing happened again, and again the robot was on for an hour or so when it happened. This time, it took a CAN ID change and a couple of reboots to get it working again.

Third time today and the CAN ID change and the rebooting does not help. Upon starting robotInterface I get the message ``[ERROR]sharedcan [0] board 6 ABS ENCODER ERROR AXIS 0``. The robot starts but the pitch axis of the torso is not functioning. 

Also, the worst was not used much (except the startup phase) the past months. In fact, it was not used at all the past 2 months.

I hope that it is due to some loose connection to the encoder somewhere (in that case, I would be grateful for some instructions on how to check them) and that a spare is not needed :)

super-ste, 2015-10-23:
Hi @kt10aan,

@julijenv is OOTO and will be back on Wednesday.
In hte meanwhile can you have a look @ his suggestion here
https://github.com/robotology/icub-support/issues/57#issuecomment-142975228
on how to check whether the cable got pinched?

kt10aan, 2015-10-26:
Update:
I was trying again today to start the robot, but the same issue was persisting. I took out the "waist" covers, and visually inspected the cable. As far as I could see, the cable was not damaged anywhere severely. However, my view of the cable was limited and I didn't know which plate to remove to get better access to it (I assume one of the back ones?)

However, moving the cables and moving the torso all the way back and all the way forth, in my attempt to get a better view, did the trick. A subsequent attempt to start the robot was successful. Also ``yarpmotorgui`` was reporting all three torso joints in good health and the torso could be controlled.

So, I will happily assume that the actual encoder board for torso joint 2 is in good health, and either the cable or the connector to it is flakey.

Is it relatively easy to check the actual connections? And what steps should I follow?

julijenv, 2015-10-29:
Hi @kt10aan ,
Is everything ok now? try to push all the different connector you can see from the left and the right side of the bottom of icub , looks like one got a bit unplugged or something (please do that when the motors ARE OFF),
Let us know,
best regards,

kt10aan, 2015-10-30:
Further update: The torso is still mostly fine. One in a while, it will give the same error. Fiddling with the  waist side plastic covers (the two small ones), usually fixes the problem. Could you please post a photo of the connectors to check?

julijenv, 2015-11-04:
Hi @kt10aan ,
please remove the 2 plastic side cover and make some pictures to understand where you may have touched and advised you on what to check to d not have this problem again.
let us knlow

kt10aan, 2015-12-15:
So, it happened again and a light movement of the covers hasn't fix it. I checked with canLoader all the Net x and everything was in order, ID-wise. 
_Except_ Net 0 [head and torso] . Sometimes it would connect, sometimes no. When it was connecting, it was reporting the two head boards having the same ID (e.g. both 1). When I was changing one of them to 2, the other one was changing to 2 too.

Since the boards connect sometimes, it must be (I hope!) some connector somewhere.
Where are the connectors for the head/torso?

Also notice please that board 5 and 7 give a MAIS error. I do not know if is a effects of the torso board.

Finally I have checked the 3 stacked boards above the elbow  in both arms as you told me in another issue and they all have their lights on etc

Here is the robotInterface log:
https://gist.github.com/kt10aan/60c74587803d2dd4e545
Also I have photos: https://db.tt/UOicSRiF (its a zip file with 14 photos or so).

kt10aan, 2015-12-16:
Another update: When i can connect to Net0 (head./torso) i get this strange image:
https://dl.dropboxusercontent.com/u/22756225/canLoaderImgOsaka01

Strange because I should have 2 x 4DCand 2x BLL, with different IDs. Instead now I have  1x 4DC and 3BLL. And the ID of the 4DC is always the same as one of the BLL, even if I change it to something else.

kt10aan, 2015-12-17:
Further update: I can connect to Net0 though the canLoader GUI but robotInterface still fails with the same log as above. If I run though the updateFirmware.sh script, canLoader reports that it cannot find the two boards that have the same id.

kt10aan, 2015-12-17:
Further update: The only board I cannot see is the 0B0, in Can 0. However, if it is the one in the chest as the schematics show, it is powered, with lights on etc.

julijenv, 2015-12-18:
Hi @kt10aan ,
Ok I re-read everything from the beginning and I think you run into not just one but more than 1 problem.
First of all you may consider VERY useful and recommended to not simply change the id on the board which got swapped but also reload the firmware on the failing board.
(in your case , the BLL ID 2 from your screenshot is WRONG it should be 1).
So in my point of view, you should remove the face cover and unconnect the 4 pin connector right above the nose pin in order to get rid off the board which has the same ID of the faulty board, it will be simpler to detect it that way , dont forget to replug it as soon as the ID and the firmware will be download on the BLL of the neck.
IF you did so you should be able to see with canLoader later all the board on CAN line 0.
1: BLL
2: M4DC
5 and 6 : BLL (more detail from this page http://wiki.icub.org/wiki/Can_addresses_and_associated_firmware#CFW2CAN0:_Head_and_Torso_3)
About the pitch problem you run into, could you please try to find out the connector end of this sensor  that you can see on the 2 following pictures and have a check on the integrity of the wires inside the connector plug into the board. You may think about recrimp every single cable if you cant find out the weak wire. A question pop up into my mind when I m typing, do you have the crimps and crimper tool to do this job?
[image]
[image]
Let us know,
best regards,
Merry Xmas

kt10aan, 2015-12-18:
Thanks @julijenv ! We have a selection of crimps and a crimper here, so hopefully I will have the corresponding one. I will try it on Monday morning. Merry Xmas!

kt10aan, 2015-12-21:
So, even if I disconnect the M4DC, the 0B0 BLL still cannot be found by canLoader. I run the command line canLoader with boardID=1 for parameter and still could not be found. The same if I was giving it ID 2.

All lights are on on the 0B0 BLL though: On the left side, the blue one, on the rights side yellow red orange.  Is it  a faulty board or just a faulty connection somewhere?

kt10aan, 2015-12-21:
Also, canLoader reports this (with 4DC connected):

``[INFO]CONNECTED: 3 Boards found on bus 1``
``[DEBUG]    id   type     version``
``[DEBUG]    5       4     2.84.83``
``[DEBUG]    6       4     2.82.83``
``[DEBUG]    2       3     2.21.83``

Shouldn't the bus number be 0 instead of 1?

kt10aan, 2015-12-21:
And, after checking (as far as I could see) the connections of the 0B0 board, I found this:
![Damaged cable] (https://dl.dropboxusercontent.com/u/22756225/Osaka01_0B0_cable.JPG)

Could this be the culprit ? It certainly doesn't look good. And to fix it, should I go for replacing the whole cable tape, or just repair the damaged cables?

kt10aan, 2015-12-21:
Ok, further update: The 2 outer cables from the cable tape that are damaged, don't end up in a connector, but on the air,  with some plastic tape holding them from being naked.

julijenv, 2016-01-04:
Hi @kt10aan ,
please come and join me for a skype meeting call in order to figure out the problem.
our skype ID is icub.support
C U around

kt10aan, 2016-01-06:
After our skype meeting (and some luck) all things related to head and torso work fine :) So this issue can be closed.

kt10aan, 2016-01-16:
To describe how the issues were solved:

1) The original issue with the torso was fixed as per @julijenv suggestion, by tracing the connector of the sensor in the photos above, and finding and pushing back in that connector in its socket, in the waist.

2) The second problem, which was different indeed, was more tricky. While we checked all connections of the BLL for continuity, and while we tried to unplug all other boards on that specific CAN (0), the BLL would not show up.

The N0 connector on the head is branching to two connectors, that they both connect to the filter. We checked the continuity from N0 to the connector _after_ the filter but it was showing to be working.

However, totally by chance, checking the continuity from the N0 connector on the head to the same connectors   _before the filter_ showed that only one of the double connectors had continuity. Swapping the place of the two connectors on the filter (see photo), made the BLL to appear, and we were able to fix the ID issue and reflash the board. Then everything worked fine.

![Filter] (https://dl.dropboxusercontent.com/u/22756225/IMG_0344.JPG )

https://dl.dropboxusercontent.com/u/22756225/IMG_0344.JPG

gsaponaro, 2016-01-16:
> However, totally by chance, checking the continuity from the N0 connector on the head to the same connectors before the filter showed that only one of the double connectors had continuity. Swapping [...]

Thanks @kt10aan for spending the time in giving these details. It can be useful for us to know.

kt10aan, 2016-01-19:
Ok, after putting everything back and everything working fine (and no screws were forgotten!), the torso issue, the ame one with the ABS ENCCODER AXIS 0 on sharedcan[0] appeared. When I opened the robot I just pushed the connector of the encoder, as per the instructions of @julijenv in the photos above. However, it seems that recrimping the connector is needed,

My question is: What is the most painless and fast way to reach in the connector, which is in the middle of the robot`s waist, and very difficulty to reach (and also to put it back)

Which parts should I takeout to have the easier and faster access?

julijenv, 2016-10-04:
Hi @kt10aan ,
could you try to sum up all the different opened issues , closed those old one, and reopen a new one with the overall issue not resolved, thx in advance

closed 2017-05-02 by kt10aan