robotology/mesh-support #92: IMU is horizontal in iCubGui

hu-yue, 2015-10-23:
This is a problem that we always had but never got solved.
In the iCubGui our IMU is pointing horizontally towards the robot's right.
I've been told several times that this is a visualization bug (that affects only our icub?), but is it possible to fix it? 
Thanks!

traversaro, 2015-10-23:
I actually investigated it some time ago, and perhaps there is an easy solution. I am now in a meeting, I will post the details later.

ale-git, 2015-10-23:
I've written the iCubGui, and if there is a bug it can be fixed, but I didn't understand what happens... can you provide more details, maybe a screenshot?

hu-yue, 2015-10-23:
Here are a couple of screenshots of the robot standing on the ground:
[image]
[image]

traversaro, 2015-10-23:
@ale-git the `iCubHeidelerg01` has an IMU that has physically a different orientation from the one on usual iCub , when the robot has all the joints set to 0.

traversaro, 2015-10-23:
That seem a bug in who is publishing the world-to-base position, namely wholeBodyDynamicsTree .

ale-git, 2015-10-23:
Yes I see, the IMU visualization can be fixed in the iCubGui configuration file... I'll take a look as soon as possible.

hu-yue, 2015-10-23:
@traversaro before the world ref frame was l_sole, now I changed it to r_sole, but nothing changed in the gui

traversaro, 2015-10-23:
@hu-yue the right foot is apparently a separate issue, see https://github.com/robotology/codyco-modules/issues/141 .

ale-git, 2015-10-26:
@hu-yue In icub-main\app\iCubGui there is the configuration file skeleton.ini, where the line
INERTIAL -34.0 82.5 0.0 90.0
specifies the D-H configuration of the inertial unit. You should play with the last two parameters (frame rotation) until you have correct IMU representation in the iCubGui. Alternatively, if let me know the IMU configuration in your torso, I can do it for you. Hope this helps.

hu-yue, 2015-10-26:
@ale-git I was able to adjust the visualization, thanks!

closed 2015-10-26 by hu-yue