robotology/mesh-support #93: push button of motors power is broken

marcoaccame, 2015-10-27:
the push button is broken and the user needs a stylus, a pen or a screwdriver to power motors on/off.
i have taken possess of the broken part so that it does not get lost.

the repair is mildly urgent because @DanielePucci needs the robot for heavy tests before external demo of end november.

DanielePucci, 2015-10-27:
Thanks @marcoaccame for noticing the issue. Actually, the power on button is pretty important :D

DanielePucci, 2015-11-13:
Guys, can we fix this ASAP?

DanielePucci, 2015-11-13:
@julijenv @super-ste

super-ste, 2015-11-27:
Done thanks to Marcello Savoldi

[image]

closed 2015-11-27 by super-ste

DanielePucci, 2015-11-30:
Very nice job. Thank you very much guys.