robotology/mesh-support #94: Tendons broken

bhigy, 2015-10-28:
Left hand - middle finger proximal tendon is broken
Right hand - pinky and little finger is broken too

spada1978, 2015-10-29:
if you could get your hands in the morning we can repair

bhigy, 2015-10-29:
Right hand repaired thanks to @spada1978. Unfortunately I have to say that the fix didn't last long. The same tendon broke again after a few grasps.

I will need the hand for my experiment tomorrow morning so please don't touch it until then.

Thanks !

super-ste, 2015-11-03:
@bhigy 
Please let us know when we can get the hand for repair.

julijenv, 2015-11-03:
Left hand medium proximal joint restored.

spada1978, 2015-11-03:
It was repaired the last joint of the right hand , and replace the cable sheaths of the passage of wiring

closed 2015-11-03 by spada1978