robotology/mesh-support #95: Instructions for replacing right ankle pitch tendon

kt10aan, 2015-10-29:
I am trying to find the pdf with instructions for replacing a broken tendon for our robot's left ankle pitch joint (j4), but with no success. 

The page at http://wiki.icub.org/wiki/Tendons_and_replacement doesn't mention anything about the legs.

julijenv, 2015-10-29:
Hi @kt10aan,
please try to get an eye at the other anckle to understand how it gets cabled, it is quite simple!
make the symetry.
if needed just come on skype: ID = icub.support

julijenv, 2015-11-04:
Hi @kt10aan ,
Did you succeed to replace the tendon?

kt10aan, 2016-01-19:
Well, after fixing the more important torso and BLL issues, its time to replace the tendon. But here we have tendons labeled from number 1 to 14, with no other explanation. Which is the one we need for the left ankle?

kt10aan, 2016-04-19:
Bump :), I still have no idea which tendon to use. As I said, we have tendons labeled from 1 to 14. Is the left ankle pitch tendon one of these? Or do we lack the corresponding tendon?

julijenv, 2016-04-19:
Hi @kt10aan,
From memory it is the cable 14 but if you still have the broken cable you can try to see if the length make sense comparing them.
Afterwards just take a look at the other ankle and mirror the cabling. 
Let us know

kt10aan, 2016-04-20:
The cabling pattern seems very simple. However, cable 14 is too long compared to the one that broke.

So there are two possibilities:

- The correct cable is not the 14
- We have wrongly labelled spare cables

Also: the round metal part at the middle of the lower leg (inner side) has to be move so the groove is in the horizontal position (as it is in the right leg).

[image]

What is the procedure to do so? I assume it is the same as for the shoulder (loose a screw, then insert two metal rods)

julijenv, 2016-04-22:
Hi @kt10aan ,
I was wrong saying the number of the cable was 14.
The number of this cable is 16, we will add it to your request #225 .

julijenv, 2016-10-04:
Did yuou fix the ankle joint?
@kt10aan

kt10aan, 2016-12-16:
Hi @julijenv,

as we received the spare tendons with a smaaaall delay, I didnt get to fixing the ankle joint until now. You were right! It was quite simple and the most enjoyable tendon replacement ever :D
Closing.

closed 2016-12-16 by kt10aan