robotology/mesh-support #96: Problems in updating linux icub-live 5 on pc104

urielmtz, 2015-10-30:
Hello,

We tried the .iso image using Debian and we have the output shown in the attached picture. We are following the steps as indicated in the wiki page but I am wondering if we are missing something else?.
Thanks.

[image]

Kind regards
Uriel

marcoaccame, 2015-10-30:
hi @mbrunettini have you got any idea about what has happened? ;-).

mbrunettini, 2015-10-30:
As described in the [wiki](http://wiki.icub.org/wiki/The_Linux_on_the_pc104#How_to_use_the_icub_iso2usb.sh_script) **you MUST use a device NOT a partition**
In the above picture is clear that A PARTITION (sdb1) has been used.
Moreover, before issuing the command be sure to use the correct device correponing to the USB device you want to write you: if you use the wrong device, you may destroy the filesystem on the PC you are using.

urielmtz, 2015-10-30:
Thanks, I replace sdb1 by sdb and the script created in the usb key the live and persistence partitions. Live partition contains the operating system and the persistence partition contains a persistence.conf file.

I put back the usb key inside the iCub head but pc104 isn't replying to the ping.

Cheers
Uriel

mbrunettini, 2015-10-30:
Which address are you trying to ping?

urielmtz, 2015-10-30:
I am doing ping 10.0.0.2 and also ping pc104

mbrunettini, 2015-10-30:
The first is correct, the second is correct only if you have an entry on the DNS.
You should connect a monitor to the PC104 to see what's wrong.

closed 2015-10-30 by mbrunettini

mbrunettini, 2015-11-04:
Issue resolved by using the icub_iso2usb.sh script with the supported linux distro.