robotology/mesh-support #97: board skin left leg

robertocalandra, 2015-10-30:
Justa  reminder that the issue with the skin on the left leg mentioned in https://github.com/robotology/icub-support/issues/14 is still open.

julijenv, 2015-11-04:
Hi @robertocalandra ,
please check  my reply on the first issue #7  an close this issue or respond onto it until we fix the robot.

robertocalandra, 2015-11-11:
I assume that you refer to issue #14, since issue #7 is head calibration of iCubParis01

francesco-romano, 2015-11-21:
I summarise here the issue on the skin board, because I think it is better to properly track this issue here, instead of the more general thread related to iCubDarmstadt01:

First report by @randaz81:

Unable to connect to legs skin. Most likely, a shortcut on one of the patches is responsible for the continuous reboot of skins connected to boards EMS10 and EMS11. I did several tests to identify the faulty skin patch, but unsuccessfully(*). This issues should be directly addressed by icub-support@iit.it

(*) on EMS10
```
can1 (front)      can1 (back)         can2
      X                                                 ---> OK
                                        X               ---> OK
                         X                              ---> CANNOT BE TESTED
      X                  X                              ---> NOT WORKING
      X                                 X               ---> NOT WORKING
      X                  X              X               ---> NOT WORKING
```

Answer by @julijenv 

Hi @robertocalandra ,
what kind of test did u perform to understand and deduce that the problem with the skin is due to patch short circuit and not just drop of voltage on can line due to to little or flaky red or black power line wires which cant bring enough current on the line?
Are they all restarting ? or just part of them?
can you remove all the connectors (8 pins) from every MTB (skin) boards and check if the MTB work all fine when the patch skins ( the 8 pin connector) are not connected to the MTB board.
Let us know

julijenv, 2015-12-18:
Hi @robertocalandra ,
Here what you will have to do, when you will have at least 2 weeks without having to use your robot because we need you to send us back the legs cover and 2 boards.
When unmounting the covers - please take as much as photos as u can in order to be able to rewire everything back and rescrew correctly evrything.
First, remove the left covers starting from the tibia you should find 10N2 wire cable connected to the cover and afterwards the 2 patch covers for mthe upper leg, each patch is connected with th 10N1 wire.
Secondly, if you follow the 10N1 and 10N2 wire they go up to the board EB10 (this board is dedicated for the skin on the left side of the femur. We need this board back as well to be able to reboost the power line on both can Line. To remove it safely and be able to use your robot later you will need to take out the E7 from the EB10 and connect it directly where gets out E6 from board EB6.
Same explanation for the right leg cover except that 10 is replaced by 11 and E7 by E10 and E6 by E9.
Send us the pack of 2 boards and the covers, and we will make the change and check the skin patches and resend everything at the speed of light.
Does it sound good to yoiu?
Let us know,

IIT - ICUB FACILITY
via morego 30, 16163 GENOVA 
ITALIA

@super-ste , @maggia80

robertocalandra, 2016-01-04:
Hi @julijenv,

can we organize a skype meeting as soon as possible so that you can follow me troughout the procedure? 

@iron76 is this reparation acceptable/influencing your CoDyCo demo schedule?

iron76, 2016-01-05:
Yes, I do not see any conflict. We can double check with @DanielePucci.

DanielePucci, 2016-01-11:
Fine for me. Let's do it ASAP.

julijenv, 2016-01-11:
Hi @robertocalandra ,
please get in touch with someone of the icub-support team (me most probably) on skype.
skype ID: icub.support
C U around

paraschos, 2016-03-24:
Hi @julijenv,

we sent the parts you requested on 23/3

Best,
Alex

julijenv, 2016-03-31:
Received part and everything is under testing since yesterday, Let u know when it will be shipped back

closed 2018-01-16 by julijenv