robotology/mesh-support #100: pc104 doesn't start in icub-cluster.py

urielmtz, 2015-11-04:
Hello,

We updated pc104 to version 5.0-15-10-26 and it replies to the ping, but when we run pc104 with icub-cluster.py its checkbox isn't ticked.

Cheers
Uriel

barbalberto, 2015-11-05:
Can you please login into pc104 through ssh and check the hostname of the pc104?

urielmtz, 2015-11-06:
The hostname in /etc/hostname is pc104, is this ok?

Cheers
Uriel

-- 
*Uriel Martinez-Hernandez*
Research Associate
Sheffield Robotics and the Department of Psychology
The University of Sheffield
Mappin Street, Sheffield, S1 3JD, United Kingdom

*Contact*
Mobile phone: (+44) 7 58 20 32 522

*Personal web page*
http://uriel-martinez.wikidot.com

*LinkedIn*
uk.linkedin.com/pub/uriel-martinez-hernandez/39/a48/a17

super-ste, 2015-11-26:
Hi all, was this issue solved? 
@barbalberto do you have further troubleshooting steps for this?

julijenv, 2016-10-04:
any news?
@barbalberto , @urielmtz

closed 2018-01-16 by julijenv