robotology/mesh-support #102: error in opening driver <xsensmtx>

marcoaccame, 2015-11-04:
Starting from yesterday, I have experienced quite often (3 times out of about 20 launches of robotInterface) that robotInterface cannot be launched because of error in opening the driver <xsensmtx>.

I powered off and then on the motors before launching robotInterface. I am however quite sure that I have waited a long enough time (all boards could bootstrap and exit the bootloader phase).

All times I launched a second time and the error did not appear anymore.
 
[image]

barbalberto, 2015-11-04:
xsensormtx is the IMU device, it is a USB device plugged into the pc104 so it is not affected by motor power off/on. Maybbe the usb connector is not well plugged in

marcoaccame, 2015-11-04:
thank you @barbalberto, I reported that just because I wanted to give as much information as possible.

Anyway, situation is getting worse and worse. Now it happens about 1 launch of robotInterface every 4, without powering off/on. 

Thus, an intervention is urgently required as @DanielePucci wants to use heavily the robot.

super-ste, 2015-11-26:
hi all,
what is the status of this?
@DanielePucci, is it there any intervention needed on our side?
cc @julijenv

julijenv, 2015-11-26:
Did see any other complaint check the wires, but nothing to say about it, @DanielePucci @francesco-romano , did u remove the sensor to didnt get the error message poping up????

francesco-romano, 2015-11-26:
I had not experienced this for at least one  week.
For me we can close and if it happens again I'll open again.

closed 2015-12-18 by super-ste