robotology/mesh-support #103: Software update for iCubGrenoble01

EliseiFrederic, 2015-11-04:
Dear IIT support,

We stopped updating the robot and repository during a few months, while 3 students were working
daily with our iCub for their internship. I tried to update our iCub contrib in July, but it prooved unsuccesful and I had to step back.

Would any of you be available to update the repositories, file structures, etc for yarp and iCub related software on our iCub laptop, using the installed TeamViewer for example ?

Thanks in advance,
Frédéric, for the Grenoble team

randaz81, 2015-11-11:
Hi Frederic, please contact me on Skype so we can schedule the update

EliseiFrederic, 2015-11-30:
We had several difficulties with the last "persistent" live distribution.

First, it seems that eth0 and eth1 are swapped on our pc104...
And the persistent distribution got things harder to debug (while booting on a laptop, to have screen and keyboard), because
/etc/udev/rules.d/70-persistent-net.rules keeped track of all the seen network cards, creating eth2, eth3 and so on...
Should we create a ticket for that, with our solutions, or are we the first to experiment this ?

Second, we cannot use the 10.0.0.1 adresses here in the lab vlan. Changing the adresses in /etc/hosts was all but enough, because several scripts/files use "10.0.0.1" directly
(for examples, /etc/rciCub.d/S90_mount-remote-fs.sh for its pings,
and /etc/rciCub.d/mounts.list for the imports)

@randaz81 , we can go on with the update from this point, if it's ok with you. Thanks in advance

EliseiFrederic, 2015-12-11:
Thanks to @randaz81  and @barbalberto, we can now run robotInterface and test with yarpmotorgui again. We just *miss the portaudio on pc104.*

Our own code/demo does compile but not run yet :-(
Marco, can you contrib *the working generalPlayer (or generalPlayer2)* to help us adapt our old code ?

Thanks again, Frédéric

EliseiFrederic, 2015-12-15:
**Concerning portaudio** : 
I did 
sudo apt-get libportaudio0 libportaudio2 portaudio19-dev
It allows yarp to be compiled with portaudio support, and "yarphear" to work on the pc104 live distrib, but some audio packets are lost...
@randaz81 : didn't you correct some bug in portaudio (sources) a long time ? they were not pushed to the (portaudio) forge, did they ?

EliseiFrederic, 2015-12-15:
**Concerning generalPlayer : **
I did some modifications to the code,
switching from VOCAB_CM_POSITION to VOCAB_CM_POSITION_DIRECT
so that
driver->ipid_ll->setReferences(ll) calls are no more rejected by the updated robotInterface
Allows our demo to run again... But is it safe to test again our arm pointing trajectories with this ?

@randaz81 : if you have an official release of generalPlayer, I'm ok to upgrade my source !

randaz81, 2015-12-27:
@EliseiFrederic, concerning Concerning generalPlayer: 
An official version of generalPlayer has been pushed to icub-main/tools, with the name trajectoryPlayer. It's not only a porting, the software has been refactored too. I tested it with the trajectories you attached to issue https://github.com/robotology/QA/issues/74 and everything works fine (no limits exceeded). I'm thus closing this issued.

IMPORTANT: If you update icub-main, you will forced to:
* update icub-firmware-shared (otherwise icub-main does not compiles)
* install the new firmware build 85 (your current version is 83, if I remember correctly). This update is mandatory.

closed 2015-12-27 by randaz81