robotology/mesh-support #98: Weight of iCub V3

DanielePucci, 2015-11-04:
This issue is intended for collecting information on the weight of iCub V3.

DanielePucci, 2015-11-04:
Waist components.
[image]
[image]
[image]

DanielePucci, 2015-11-04:
Upper body.
[image]
[image]
[image]

spada1978, 2015-11-18:
Left/Right arm
[image]
[image]

spada1978, 2015-11-20:
Tlkn Head with cover
Kg 2,69

[image]

spada1978, 2015-11-20:
Tlkn Head without cover
Kg 2,32

[image]

DanielePucci, 2015-11-25:
I just wanted to mark somewhere that @spada1978 sacrificed a finger for this issue.

spada1978, 2015-11-25:
It is for a good cause no problem

spada1978, 2015-11-25:
Upper body + wiast without battery
Kg 16,05 
photo incoming

spada1978, 2016-02-17:
buttocks
Kg 1,36
[image]

spada1978, 2016-02-17:
hip
Kg 2,96

[image]

spada1978, 2016-02-18:
Shoulder - arm - forearm - hand
Kg 3,60
[image]

spada1978, 2016-03-09:
Lowerlag
Kg 4,45
[image]

spada1978, 2016-03-09:
Lowerlag + Knee
Kg 5
[image]

closed 2018-01-16 by julijenv

fjandrad, 2018-08-29:
[image]

iCubGenova04 on the scale. 
**Total weight = 33.33kg**
@DanielePucci

DanielePucci, 2018-09-01:
@fjandrad 
I do not understand why the [comment above](https://github.com/robotology/icub-support/issues/98#issuecomment-416872933) has been put in the present issue. Please notice that [the present issue](https://github.com/robotology/icub-support/issues/98#issue-114999396) is concerned about the weight of only iCub V3, or as we call it iCub3. So, I strongly suggest to remove the comment above, and update the table https://github.com/robotology/QA/issues/61#issue-110842856 to keep trace of iCubGenova04 weight.