robotology/mesh-support #99: Mechanical adjustment of eyes needed

Tobias-Fischer, 2015-11-04:
Hi @super-ste and @julijenv,

we found that our eyes are misaligned, i.e. one eye is tilted a little further up than the other. This causes some trouble when using the stereo vision pipeline using the higher resolution of the camera. The misalignment is more than a few pixels, so I guess we need the mechanically adjust it.

Is there any documentation on how to do it around? If not, would it be possible for you to remotely help us out via Skype?

Thanks a lot,
Tobias

PS: We will host an integration meeting at the end of the month, and it would be great if we could sort this out before then :).

/cc @maxime-petit @martz21 @hjchang

julijenv, 2015-11-04:
Hi all- iCubLondon01 users ( @Tobias-Fischer , @maxime-petit , @martz21 , @hjchang )
I'll be on skype to give you an han.  Is it a recent problem or a brand new one, I mean since when you hace tis issue with theyes? what is alright at the beginning and by the use of it you got this offset, or you just for the first time use the stereo vision and realize there was a misalignemnt?
Let me know

Tobias-Fischer, 2015-11-04:
Hi @julijenv,
Thanks for getting in touch with us so quickly. Just to keep this issue up-to-date: The misalignment is 20px. We have had the misalignment for while now, but it just recently started giving us issues. Foreseeing the integration meeting, we would like to have the robot working perfectly, and this is the only remaining issue which needs to be tackled.

Thanks,
Tobi

Tobias-Fischer, 2015-11-04:
Here are the requested images of the eyes. The first image shows both eyes, the second one the right eye, and the last one the left eye of the iCub.

[image]
[image]
[image]

Tobias-Fischer, 2015-11-04:
Some more images with a ruler.

[image]
[image]
[image]
[image]
[image]

Tobias-Fischer, 2015-11-05:
Hi @julijenv, thanks again for your help yesterday. The alignment now is much better, thanks a lot. If we run into trouble again, we'll let you know, but I think the couple of pixels misalignment still left shouldn't be a big deal.

Thanks,
Tobi and @maxime-petit

closed 2015-11-05 by Tobias-Fischer