robotology/mesh-support #104: Upgrading robot iCubSheffield01 to icub-main version 1.2.17

marcoaccame, 2015-11-05:
The migration from icub-version 1.2.16 to 1.2.17 introduced on 10 sept 2015 for the ETH robots requires some assistance from @randaz81 for careful verification of new parameters.

The robot iCubSheffield01 still needs to be upgraded. 

Dear @urielmtz: details will follow on how to deal with that. 

Also consider that the problems in issue #88 are about new parameters introduces in this new icub-main version which are missing in the xml files.

marcoaccame, 2015-11-05:
The upgrade to this version requires careful steps, because **POSSIBLY WRONGLY TUNED / INCORRECT VALUES OF NEW PARAMETERS IN THE XML FILES MAY CAUSE DAMAGE TO THE ROBOT**.

For this reason we have done the upgrade of other ETH robots using remote presence of an expert, a fixed camera on the robot to see its reactions and in initial steps the **FAULT BUTTON PRESSED DOWN**. 

In order to do the updated we have two options: 

1. wait that the robot comes to Genova for the planned repair. At the same time we can also tune the new parameters.
2. arrange with @randaz81 a remote upgrade.

urielmtz, 2015-11-05:
Thanks for the details.

Could we go for option 2 and arrange with @randaz81 a remote upgrade? 
Thanks.

Cheers
Uriel

closed 2018-01-16 by julijenv