robotology/mesh-support #108: Problem in robotInterface after firmware update (git pull)

serena-ivaldi, 2015-11-11:
Hi, I am with @robertocalandra and Alex Paraschos, we are trying to update the iCub firmware in iCubDarmstadt01. After updating with ethLoader, we have version 2 build 15 on all boards. Launching the robotInterface now gives a lot of errors that we do not know. Here the log of the robotInterface:

||| configuring
||| no policy found
||| default config file specified as robotInterface.ini
||| checking [/usr/local/src/robot/icub-main/build-pc104/robotInterface.ini] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/robotInterface.ini
||| finding file [config]
||| checking [/usr/local/src/robot/icub-main/build-pc104/./icub_all.xml] (pwd)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./icub_all.xml] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./icub_all.xml
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./icub_all.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./cartesian/left_arm-cartesian.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./cartesian/right_arm-cartesian.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/motorControl/head-cfw2_can0-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/head-cfw2_can0-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/mechanicals/head-cfw2_can0-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/motorControl/head-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/motorControl/torso-ems5-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/torso-ems5-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/mechanicals/torso-ems5-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/motorControl/torso-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/motorControl/left_upper_arm-ems1-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/mechanicals/left_upper_arm-ems1-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/motorControl/left_lower_arm-ems2-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/left_lower_arm-ems2-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/mechanicals/left_lower_arm-ems2-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/motorControl/left_arm-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/motorControl/right_upper_arm-ems3-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/right_upper_arm-ems3-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/mechanicals/right_upper_arm-ems3-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/motorControl/right_lower_arm-ems4-mc.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/right_lower_arm-ems4-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/mechanicals/right_lower_arm-ems4-mec.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/motorControl/right_arm-mc_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/FT/left_arm-FT_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/FT/right_arm-FT_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/FT/right_leg-FT_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/FT/left_leg-FT_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/FT/right_foot-FT_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./wrappers/FT/left_foot-FT_wrapper.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/FT/left_upper_arm-ems1-strain.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/left_upper_arm-ems1-eln.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/pc104.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/FT/right_upper_arm-ems3-strain.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./general.xml 
[[01;32mDEBUG[00m]Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubDarmstadt01/./hardware/electronics/right_upper_arm-ems3-eln.xml 
[... cut ...]
[[01;34mINFO[00m]All actions for action level 1 of interrupt3 phase finished. 
[[01;34mINFO[00m]interrupt3 phase finished. 
[[01;33mWARNING[00m]ethResources::verifyBoardPresence() cannot reach BOARD 1 at attempt # 64 w/ timeout of 0.5 seconds 
[[01;33mWARNING[00m]ethResources::verifyBoardPresence() cannot reach BOARD 1 at attempt # 65 w/ timeout of 0.5 seconds 
[[01;33mWARNING[00m]ethResources::verifyBoardPresence() cannot reach BOARD 1 at attempt # 66 w/ timeout of 0.5 seconds 
Aborting (calling abort())...

francesco-romano, 2015-11-11:
I think your robot configuration is not updated.

Check what @randaz81 did here: https://github.com/robotology/icub-main/commit/27c8513227d13bbf54408cff512b9957de7f7b50

randaz81, 2015-11-11:
Additionally, you should check the firmware of the 2FOC boards. They probably  need to be updated as well (verifyBoardsPresence fails). And icub-firmware-shared as well!

randaz81, 2015-11-11:
icub-main has been just updated with new configuration parameters for your robot

robertocalandra, 2015-11-12:
@francesco-romano what does it means robot configuration?

yesterday we updated:
-icub-firmware-shared
-icub-firmware
-yarp
-icub-main

@randaz81 ok, we will update again icub-main. What is the procedure to update the firmware of the 2FOC boards?

francesco-romano, 2015-11-12:
@robertocalandra I meant what @randaz81 has just done :smile:

marcoaccame, 2015-11-12:
Hi @robertocalandra,

I will help you to upgrade the FW of CAN boards on an ETH robot.

here are the steps. later on we shall add a dedicated issue on github so that everybody can refer to it.

1. Power motors ON.
2. Launch `ethLoader` and send a *Discovery*. You repeat the command until you can see the ETH boards in updater mode. You can close `ethLoader` now.
3. Now you launch `canLoader`. In the second selection box, instead of the default *ecan* you select *EMS*.
4. Now you select the board address you want to reach and the target CAN bus. For example to see the 2FOC boards on EB1 you choose *Can1* on *10.0.1.1*.  
5. After you have selected the IP address and the CAN bus you press *Connect* and all boards on the CAN bus will appear with their FW version.
6. To update the FW you must first select the binary you want to load, the CAN board(s) on the list and then press *Start Download*.
7. At the end of the FW update the `canLoader` will show the FW version of the new application.
8. After you have finished operating on `canLoader`, before you can use robotInterface it is important to restart the ETH boards, so that they move away from the updater mode. Quickest way to do it is to power motors OFF and then ON again.

robertocalandra, 2015-11-12:
thanks, I will now proceed with the updating.
Where do we find the binaries to upload?

marcoaccame, 2015-11-12:
For upgrading to icub-main version 1.2.17 it is enough to update the FW of the EMS boards (ems.hex) and of the 2FOC boards (2FOC.hex), which can be found in icub-firmware-build.

I would however recommend also updating FW of MTB boards (skin.hex).

Moreover, **VERY IMPORTANT THING**: after the full SW and FW update **DO NOT run robotInterface straight away unless with the EXTERNAL FAULT pressed**.

Read issue #104 for further details and arrange with @randaz81 some remote assistance for fine tuning of new parameters.

marcoaccame, 2015-11-12:
@robertocalandra: you need to be sure to have pulled latest icub-firmware-build repository. from your previous post it seems that you forgot to do this.

serena-ivaldi, 2015-11-12:
I am downstairs trying to follow the procedure. And will keep you updated.

<critical mode on>
May I say that following github issues is really _not_ practical, and it was way better when there was a simple manual on the wiki?
Maybe in IIT you like being evaluated on the number of issues created/solved, but this simply reveals the need of a proper manual to follow. This procedure of referring to other issues is crazy, and really not practical. Browsing issues to find similar problems not practical if you are talking about standard procedures, this can be acceptable for code bugs and coding questions, but it is inappropriate for something that is a classical procedure for a manual. 
<critical mode off>

serena-ivaldi, 2015-11-12:
@marcoaccame I just git pulled all the folders, and icub-firmware-build was properly pulled yesterday.

serena-ivaldi, 2015-11-12:
@marcoaccame 
I just git pulled yarp, icub-main, icub-firmwars-shared, icub-firmware-build.
I recompiled yarp and icub-main on the pc104, using the build-pc104.
Everything seems to work... only that if I launch ethLoader on the pc104 now it cannot find find yarp libraries. yarp is working though, and other programs are working.
See below

icub@pc104:~$ ethLoader 
ethLoader: error while loading shared libraries: libYARP_OS.so.1: cannot open shared object file: No such file or directory
icub@pc104:~$ yarp
This is the YARP network companion.
Call with the argument "help" to see a list of ways to use this program.

randaz81, 2015-11-12:
ethLoader was not built correctly.
1) check that you are not using old executables with new libraries (`which ethLoader` ecc)
2) maybe yarp/icub-main was built with old cmake options? Keep a backup copy of your build folders and start cmake in a new folder, with a clean cache.

serena-ivaldi, 2015-11-12:
I re-did it in the meantime, now the error is:
icub@pc104:~$ ethLoader 
ethLoader: error while loading shared libraries: libethLoaderLib.so: cannot open shared object file: No such file or directory

serena-ivaldi, 2015-11-12:
I have an idea: can you confirm that on the pc104 you do not have to do make install? maybe this is the problem.
I checked in their bashrc they have 

export PATH=$PATH:$ICUB_DIR/bin:$YARP_DIR/bin

but there ethLoader is in 

icub@pc104:~$ which ethLoader
/usr/local/bin/ethLoader

It's an old file and my recompiling doesn't overwrite it because there have been no changes since then. Could it be that?

randaz81, 2015-11-12:
It is that. Do make install to replace the wrong old executable with the new one.

lornat75, 2015-11-12:
Actually if the PATH points to $YARP_DIR/bin do not do install.

Do:

sudo make uninstall

Double check that the yarp+icub executables are not in /usr/local/bin

then do simply:

make

lornat75, 2015-11-12:
@serena-ivaldi: about your complain. We use github issues so that we can better track people's requests. It may not be optimal but it is much better than using emails (issues are public and can be useful to others, they are indexed by google, you can reference other issues, you can add people to the conversation and you can search using keyworkds, labels etc..).  

Issues do no substitute procedures, apologies if the procedure for updating the firmware is not in the wiki we will add it!

serena-ivaldi, 2015-11-12:
@lornat75 Thanks :) I think it definitely needs to be in the manual!!

lornat75, 2015-11-12:
@marcoaccame, @randaz81 Can you have a look and update this page: http://wiki.icub.org/wiki/Firmware

serena-ivaldi, 2015-11-12:
@lornat75 @randaz81 
the uninstall worked! now i can launch ethLoader.
i will try to follow the firmware update instructions of @marcoaccame

serena-ivaldi, 2015-11-12:
@marcoaccame I launched ethLoader and clicked discovery. how do I know when the boards are in updater mode? there is apparently nothing changing.
I can see 11 boards with IPs 10.0.1.1-11, vers 2 build 17.
[image]

robertocalandra, 2015-11-12:
if you see 2.17 they are in update mode.

serena-ivaldi, 2015-11-12:
@marcoaccame Sorry for bothering but it's the first time I do this update and I am afraid of making mistakes!

I tried updating the firmware starting from the 1st board.
IP 10.0.1.1 + EMS + CAN 1
there are 4 boards 2FOC, saying release 1, version 3D, build 0
I selected all the 4 boards, and updated file 2FOC.hex
now I can see 4boards 2FOC, saying release 3, version 0, build 0

Is that right?
What am I flashing precisely? Because there is no " additional note" in the firmware, I remember in the old CAN system every board had the name of its body part (e.g., right arm)

Please confirm that I am doing right... I will continue flashing the other IPs

marcoaccame, 2015-11-12:
maybe we should open a dedicated issue on github ;-).

marcoaccame, 2015-11-12:
just kidding.

serena-ivaldi, 2015-11-12:
i am not laughing

serena-ivaldi, 2015-11-12:
@marcoaccame just to check, i am now on 10.0.1.2 and I can see 4 boards:
- MAIS
- RM 4DC

Is the mais supposed to be updated with skin.hex?
What about the other RM 4DC boards?

marcoaccame, 2015-11-12:
wait a minute.

marcoaccame, 2015-11-12:
first thing: the 2FOC.

it is correct that you have fw version 3.0.0, as you can read from the git log of 2FOC.hex

serena-ivaldi, 2015-11-12:
ok i am flashing the 2FOC also in the other IPs.

what about the other boards?
i can see MAIS, strain, RM...

also should i do can1 and can2? what is the difference?

marcoaccame, 2015-11-12:
then: i have already written that it is necessary to update the 2FOC only.

Possibly also the skin boards.

the 2FOC boards are always on CAN1 bus on addresses: 10.0.1.[3, 5, 6, 7, 8, 9] 

the skin boards are on CAN2 of addresses 10.0.1.[2, 4] and on CAN1 / CAN2 of 10.0.1.[10, 11].

serena-ivaldi, 2015-11-12:
yes but what is possibly? of course it is possible, i am updating. so i will do it.

what is the name of the boards for the skin? I remember MAIS.
Are strain and RM also connected to the skin?

serena-ivaldi, 2015-11-12:
we just flashed the skin boards on 10.0.1.2, CAN2.
The canLoader was saying Release 2, vers 10, build 17.
But the debug information was saying that it discivered version 2.16.17
--> i think there is a bug somewhere

Anyway, we uploaded the skin.hex new version (git says 2.17.0)
Again, the canLoader shows release 2, version 11, build 0.
However the debug information says it successfully uploaded version 2/17.0
 
We are going on uploading the other IPs for the skin

marcoaccame, 2015-11-12:
I suggest that you to update all 2FOC first. That is the minimum you need to have the system ready for the new version of icub-main. 

Then, if you have time you can also update the skin boards. They have new features that you can read in the log of skin.hex. The name of skin boards is not MAIS. It is skin. You can see them in 10.0.1.[2, 4] CAN2

All other boards (MC4, STRAIN, MAIS) dont need to be updated.

marcoaccame, 2015-11-12:
The canLoader uses hex notation, the debug information decimal.

serena-ivaldi, 2015-11-12:
We have updated everything (2FOC first then the SKIN). 

However we have a problem with the boards 10.0.1.11 and 10.0.1.10. 
They do not answer to canLoader (Communication Error, No answer received - no boards found), both with CAN1 and CAN2.

Since iCubDarmstadt is now currently without the upper cover of the left leg, so this could be related. However it is 2 IPs and 2 CAN bus... is that the problem? Apparently @robertocalandra and the other guys removed it because of some hardware errors. Should we put it back?

Could you confirm that the 2 IPs are related to the left lef? This would help debugging. In the meantime I will try to turn off and on the boards.

marcoaccame, 2015-11-12:
board 10.0.1.10 is on left leg. it has attached two patches of skin. one on can1 and one on can2.

board 10.0.1.11 is the same but for right leg.

if the canLoader gets such an error it typically means that there are not CAN boards attached, or a wire is broken.

serena-ivaldi, 2015-11-12:
I found a document about the placement of the boards here:
http://wiki.icub.org/wiki/Electrical_wiring#Logic_and_Harness_iCub_2.5-_ETHERNET_based

It seems the board 10.0.1.11 is in the right leg, while 10.0.1.10 is in left leg.

So we can say it is normal for the left leg (no upper cover of the skin right now) but there is a problem in the right leg. How can we debug if there is a broken cable on the right leg?

Also, @robertocalandra was reporting power issues on the legs, could be related to that?

marcoaccame, 2015-11-12:
now @serena-ivaldi i need to go because i am on a meeting. 

if you need immediate help, please arrange a skype session with someone from icub-support.

serena-ivaldi, 2015-11-12:
ok thanks for helping. we will leave the right leg skin not updated - not a big issue right now.

pattacini, 2015-11-12:
Let me chime in and moderate this exchange.

We try to organize skype/remote sessions to help people having whatever issues with the robot and/or the software. Please, let's stick to this policy and avoid using this service as a chat.

serena-ivaldi, 2015-11-12:
so we are launchin robotInterface now and of course there is plenty of errors because it cannot connect to boards 10 and 11. should I remove the corresponding items in the config file of the robotInterface?

another problem: we are trying to debug with yarplogger but it does not show anything. is it normal?

serena-ivaldi, 2015-11-12:
@pattacini can you come on skype now? sorry to put pressur but i am staying only 1 day basically and i would like to fix it before leaving.

I commented the two lines of the skin in the XML file of the robotInterface.ini

So the robotInterface does not seem to bother anymore about that. However, it's giving errors and the robot won't start or calibrate. Here is the error from the robotInterface:

[INFO]created device <parametricCalibratorEth>. See C++ class yarp::dev::parametricCalibratorEth for documentation.
[ERROR]Head_Calibrator : invalid number of calibrationZero params 
yarpdev: ***ERROR*** driver <parametricCalibratorEth> was found but could not open
[WARNING]Cannot open device head_calibrator 
[WARNING]Cannot open device head_calibrator 
[WARNING]There was some problem opening one or more devices. Please check the log and your configuration 
[ERROR]One or more devices failed opening... see previous log messages for more info 
[DEBUG]ethResources::goToConfig() called for BOARD 5 
MANAGEMENT-appl-status: sign = 0xaa000000, board EB5 -> name = aEMS00v2mec
                        version = 2.15, built on date 10 11 2015, at hour 18:15
                        control loop timings: RX->400us, DO->300us, TX->300us
                        state = applstate_running
MANAGEMENT-appl-status: sign = 0xaa000000, board EB5 -> name = aEMS00v2mec
                        version = 2.15, built on date 10 11 2015, at hour 18:15
                        state = applstate_config

pattacini, 2015-11-12:
Nope, sorry

/cc @robotology/icub-support-team

randaz81, 2015-11-12:
Head calibrator issue fixed. Please update icub-main.

francesco-romano, 2015-11-12:
@serena-ivaldi @robertocalandra if there are any issues on hardware or electronic level please contact ASAP @robotology/icub-support-team and arrange a Skype meeting, in order to assess the gravity of the problem.
We need to know urgently the state of the robot.

cc @DanielePucci @lornat75 @iron76

serena-ivaldi, 2015-11-12:
@francesco-romano I am trying to fix the issues, stay tuned!

@randaz81 I gut pulled and the error on the head disappeared. Thanks! Now it is calibrating. I still have two issues:

1) The error on the boards (10 and 11) .. not possible to debug the robotIntefrace with these errors, there's red everywhere! Is there a way to disable the creation of the drivers for these two boards? I tried commenting the xml file in the lines related to left and right leg skin, but it is not sufficient apparently. 

2) The calibration of the right arm does not look super correct. It is calibrating the left arm then the right one. But the lower arm is not moving symmetrically as the left (a thin difference), and the wrist is down (problem in the zero calibration?). I also heard a bad noise (like something sudden moving a cable) coming from the right wrist. 
Do I need to redo the wrist calibration?

DanielePucci, 2015-11-12:
In relation to 
~~~
Maybe in IIT you like being evaluated on the number of issues created/solved [...] 
~~~
I can only say [1]
~~~
[...] let us not talk of them, but look and pass.
~~~

[1]: Divine comedy, III, 51 http://www.poetryintranslation.com/PITBR/Italian/DantInf1to7.htm#anchor_Toc64090919

marcoaccame, 2015-11-12:
@serena-ivaldi, if you comment out from robotInterfce.ini down to whatever it calls any reference to the skin on legs, then boards 10 and 11 are not used at all. It may be that you edited a file and robotInterface is using another one.

In short, if your robotInterface.ini is as in repository which calls icub_all.xml, then to remove presence of boards 10 and 11 you should use the following extract of icub_all.xml  

```xml
    <!--  SKINS -->
    <devices file="wrappers/skin/left_arm-skin_wrapper.xml" />
    <devices file="wrappers/skin/right_arm-skin_wrapper.xml" />  
    <devices file="wrappers/skin/torso-skin_wrapper.xml" />  
    <devices file="hardware/skin/torso-cfw2_can9-skin.xml" />
    <devices file="hardware/skin/left_arm-ems2-skin.xml" />
    <devices file="hardware/skin/right_arm-ems4-skin.xml" />

   <!-- removed because we don't want skin in legs
    <devices file="wrappers/skin/right_leg-skin_wrapper.xml" />
    <devices file="wrappers/skin/left_leg-skin_wrapper.xml" />
    <devices file="hardware/skin/left_leg-ems10-skin.xml" />
    <devices file="hardware/skin/right_leg-ems11-skin.xml" />
    -->

```

serena-ivaldi, 2015-11-16:
Hi guys, just to update, I commented the entries of the legs skin in the icub_all.xml
Together with @julijenv @robotology/icub-support-team we also fixed the issue in the wrist calibration and updated the calibration files.
Now the robot calibrates and the robotInterface runs ok (no message errors).

We created (thanks to Alex Paraschos)  a pull request to update the files on the repository, please accept it!

closed 2015-11-16 by robertocalandra

jeljaik, 2016-06-03:
Hello guys, this is just a reminder that the instructions for firmware update on ETH robots is still not clearly available in iCub's wiki. If it is at least, not in the firmware update section. Thanks!

hu-yue, 2016-06-03:
@jeljaik I don't know if it's in the wiki, but I know there's a howto file in the `icub-firmware-shared` repo at least for the `EMS` : https://github.com/robotology/icub-firmware-build/blob/master/EMS/howto.txt

jeljaik, 2016-06-03:
You’re the best. Thanks Yue! I was forgetting about it.