robotology/mesh-support #114: Problems setting up newly upgraded Plymouth iCub

svadams, 2015-11-18:
Dear iCub experts,

We are in the process of setting up our newly upgraded iCub but having some difficulty. There are not many people here who remember how we did it last time and no doubt things have changed!

The first main problem seems to be that our host pc does not recognise the hostname of the laptop (icubsrv). I can ssh in using icub@10.0.0.1, and then run yarpserver etc. but cannot do this all in one go like we used to i.e. with ssh icub@10.0.0.1 "yarpserver".

I have looked through the instructions for setting up icub laptop and it appears most things were done for us, but maybe there is something we are missing?

If anyone has any suggestions for things we should try or check we'd be grateful!

Regards
Sam

julijenv, 2015-11-19:
Hi iCubPlymouth01 users,
The new robots getting out of our facility for upgrade get a brand new version fo the pc104 usb key and this one unfortunately has the password set as : live.
please follow steps on the copy certification passwrod ssh key on 
http://wiki.icub.org/wiki/The_Linux_on_the_pc104
[GitHub](http://wiki.icub.org/wiki/The_Linux_on_the_pc104)

mbrunettini, 2015-11-19:
In details : 
* http://wiki.icub.org/wiki/The_Linux_on_the_pc104#Users_and_Passwords
* http://wiki.icub.org/wiki/The_Linux_on_the_pc104#SSH_passwordless_login
* http://wiki.icub.org/wiki/The_Linux_on_the_pc104#Common_Problems_and_how_to_solve_them

Let us know.

svadams, 2015-11-20:
Hi guys,

Thanks for your responses! I have fixed the automatic ssh login.
However, the main problem we still have is that our host pc does not recognise the actual hostnames of the pc104 or the laptop (icubsrv) - we have to use the IPs 10.0.0.1 and 10.0.0.2 to login. This isn't a huge issue but would rather have everything setup properly.

Is this likely to be something we need to do on the host or the laptop?

Regards
Sam

svadams, 2015-11-20:
Hi again,

I fixed the problem now - was just a case of adding the hostnames to /etc/hosts on our pc!

Sorry to bother you!

Regards
Sam

mbrunettini, 2015-11-20:
good, issue closed.

closed 2015-11-20 by mbrunettini