robotology/mesh-support #117: Right hand's proximal joint is broken

nawidj, 2015-11-19:


pattacini, 2015-11-20:
We need also to: 
- [ ] replace the hall sensor of the same part.
- [ ] re-crimp the CAN cable of the torso skin.

/cc @julijenv

super-ste, 2015-11-20:
@nawidj can yo uplease add a short description of what broke the tendon (e.g. testing new code, accidental collision with environment, normal use, electronic failure [this looks like to be the cause in our case] etc etc)

@pattacini shall we open a new issue for the skin?

pattacini, 2015-11-20:
@julijenv fix these

pattacini, 2015-11-20:
However, in light of #111, I'd catch the chance to stress that it is really important to document any request.

closed 2015-11-20 by pattacini

nawidj, 2015-11-20:
This was my first request, future maintenance request will be documented properly. The problem was pointed out by @julijenv, I did not know exactly what caused the problem and whether it was a preexisting problem.

@pattacini the problem noticed by @julijenv was related to the right index finger's distal joint. Has that been fixed or should I open a new maintenance request?

pattacini, 2015-11-20:
@nawidj the problem has been solved, thanks, no need for new issue.