robotology/mesh-support #119: cfw002 kernel driver for Debian 8/Ubuntu 14.04

plinioMoreno, 2015-11-20:
Vizzy is running the current version of the cfw002 kernel driver, that works properly on Ubuntu 12.04. We want to upgrade to the newest LTS (14.04), and we would like to know if you have scheduled a new image that works with Debian Jessie (8), that should work as well on Ubuntu 14.04.

cc @lornat75 @mbrunettini @barbalberto

mbrunettini, 2015-11-23:
Not at the moment: unfortunately Debian changed the live image scripts from v7 to v8 and old script does not work anymore in Jessie.

Tobias-Fischer, 2016-02-12:
Keeping this issue open as Jessie will need to be supported at some stage.

/cc @lornat75 @pattacini @gsaponaro @mbrunettini @barbalberto @traversaro

mbrunettini, 2016-02-12:
Jessie LIve image is under testing

plinioMoreno, 2016-02-12:
@mbrunettini, 
Please let me know when you commit the changes of the cfw002 kernel driver in the repository (is this [one](https://svn.code.sf.net/p/robotcub/code/trunk/iCub/pc104/device-drivers/cfw002/) ?)

mbrunettini, 2016-02-12:
Maybe this is a misunderstanding: I'm not making any changes to the kernel driver, because I'm not it's mantainer. I just added it to the live image under testing.

plinioMoreno, 2016-02-12:
@mbrunettini 

Please let me know who did the changes to the cfw002 kernel module in order to compile in Jessie. I would like to have access to the new kernel module, because we need to compile it on the special setup of Vizzy.

mbrunettini, 2016-02-12:
No changes have been made to the kernel module.
The module will be loaded into the same kernel as LIVE v5 because no RT flavour kerner is available in Debian Jessie but real time kernel is a mandatory requirement.

plinioMoreno, 2016-02-26:
I was able to compile the cfw002 kernel module on Ubuntu 14.04, kernel 3.16.0-50-generic. By using the files available at the [svn repository](https://svn.code.sf.net/p/robotcub/code/trunk/iCub/pc104/device-drivers/cfw002/src), I got into compilation errors. But after some searches on the internet, I applied a fix that is just to remove all the instances of ```__devinit``` , ```__devexit``` and ```__devinitdata```. 

After compiling and running the kernel module, I typed ```dmesg``` and got the following
```
[  179.294512] cfw002: module verification failed: signature and/or  required key missing - tainting kernel
```
Otherwise everything worked properly and I was able to connect to the control boards using ```canLoader```

@lornat75 @mbrunettini @barbalberto please cc to more people that may be interested on this information

closed 2016-02-26 by plinioMoreno