robotology/mesh-support #120: Report of on-site visit of 17/11-20/11 to Darmstadt

francesco-romano, 2015-11-21:
This report contains the activities carried out on the iCub in Darmstadt during the visit from the 17th to the 20th of November 2015.

##### Tasks performed
- Setup `yarpmanager` applications for `RobotStartup`, `FaceExpressions` and `TorqueBalancingTwoFeet`.
- Updated `icub-main`, `icub-firmware-shared` and `firmware` of EMS and STRAIN at the latest build (19th of November 2015 - https://github.com/robotology/icub-firmware-build/commit/5543553bda97619b1d64edf7010283fb4038af15)
- Robot ''fine'' calibration. The robot was heavily miscalibrated.
- Updated offset of FT sensors on legs and feet.
- Setup of iCub tests infrastructure and launched the `Optical Encoder Consistency` test.
- Motor parameters identification (the identified dataset has been uploaded in [db.codyco.eu](https://db.codyco.eu/#/experiments/564da53f73802afb0a7c0837/))
- Tighten tendons of shoulders and torso (see pending tasks of https://github.com/robotology/icub-support/issues/14)
- Torque control parameters tuned for all joints.
- Configured `yarpWholeBodyInterface` and `wholeBodyDynamicsTree` for the robot.
- Created gains files for `torqueBalancing` two feet and left foot.
- Created reference files for `torqueBalancingReferenceGenerator` for sinusoidal movement of the Centre of Mass.

The following gif files show the performances of torque balancing in two feet, two feet with com trajectory and left foot scenarios
[image]
[image]

#### Notes:
- Skin under feet has been removed to perform proper fine calibration and one foot balancing.
- Joint limits for hip roll joints has been *relaxed*. Greater attention should be paid when the robot is mounted on the pole, or also hanging from the crane to avoid that both legs "closes" at the same time. Damage can occur to both motors and covers.
- Some torque control parameters (namely `kbemf`) on the legs has been zeroed in the committed robot configuration files. This is necessary for some kind of demos (like torqueBalancing). If you need to restore the original values take a look at this commit (https://github.com/robotology/icub-main/commit/eca4b55d2c59140b9d953736902349f6873bda2b)

#### Pending tasks
- The `Balancing on Left Foot` app should be created. It can be duplicated from the two feet one, which you can find in `~/.local/share/yarp/applications` and modified accordingly. Can @robertocalandra or @paraschos take care of this?
- Put the left upper leg cover back to avoid possible damages to boards and cables (@robertocalandra).
- Investigate why the skin on the leg has not-identified problems since the first visit of @randaz81 (cc @julijenv). The discussion must be carried out in https://github.com/robotology/icub-support/issues/97.
- We observed discontinuities in the references when the torque balancing app starts. This phenomenon is very similar to what we observed before the fix at the firmware level by @ale-git in https://github.com/robotology/icub-firmware/commit/23a4b700509aa346f84dd5722e1aedade24f5d3b, but the firmware has been updated. We should investigate this issue.
- Some of the tasks in https://github.com/robotology/icub-support/issues/14 are still pending (from IIT side), and some notes (on darmstadt side) were still present during this visit. I'll close the other issue anyway.

#### Notes for Y3 Demo
- Power supply cable is too short. We need an extension
- Small room

cc @robertocalandra @paraschos @rueckert
cc @lornat75  @iron76  @randaz81  @DanielePucci  @jeljaik

DanielePucci, 2015-11-21:
If I remember well, @francesco-romano and @jeljaik also weighted the robot. So, do not forget to update https://github.com/robotology/QA/issues/61

robertocalandra, 2015-11-26:
I mounted (few days ago) the left upper leg cover back (but it is not connected)

I have no idea what you mean with creating The Balancing on Left Foot app and I also don't know how to modify it accordingly. Could you elaborate?

traversaro, 2015-11-26:
We usually use the [yarpmanager](http://www.yarp.it/yarpmanager.html) to launch multiple modules from a graphical interface and perform the necessary port connection, to avoid having to remember all the right steps to perform to launch a setup and the relative configuration of each module. 

A group of executables started by `yarpmanager` is called `application`, and is usually stored as an xml file, following the format defined in [http://www.yarp.it/yarpmanager.html#application](http://www.yarp.it/yarpmanager.html#application). 

To launch the balancing demo, several executables should be launched at the same time. An application with the right configurations for the two feet balancing is already present `~/.local/share/yarp/applications`. Create a new one for left foot balancing (copying most of the its content from the two feet balancing one, probably just some configuration parameter should change) may be a good idea to easily launch the left foot balancing demo on your setup.

francesco-romano, 2016-05-04:
cc @DanielePucci

closed 2018-01-16 by julijenv