robotology/mesh-support #121: Report after IIT@intesa SanPaolo in Torino

vtikha, 2015-11-23:
After a week in Torino I wanted to report a couple of important problems:
 - Head went in FAULT every day doing simple red-ball demo
 - Torso went in FAULT a few times, during the week, making the robot fall forwards
This is important and needs to be checked and resolved.

julijenv, 2015-11-23:
Please give more details or log files in order to let us understand the above HW FAULT source.
AFAI recall, it has been implement thePWM fix on the head so it would be interesting to apprehend the HW FAULT head issue more thoroughely before extending the above fix on the other platforms.

pattacini, 2015-11-23:
@julijenv we will have to apply the change detailed in #75 straightaway, regardless if it has or not an impact on these `HW Faults`, because the **PWM limit** must reflect the way those motors are controlled.

super-ste, 2015-12-15:
hi all, shall we close this issue as we keep #75 opened? do we have any further hints on what might have caused the torso to fall forward?

vtikha, 2015-12-16:
@super-ste let's keep the #75 opened as the head is still producing some `HW Faults`. @apaikan has made a stress test for the head and will report soon. As for this issue I would also keep it opened as it is getting worst but @julijenv is now going to re-crimp connectors on the back of the iCub. Will report any findings here.

traversaro, 2016-11-17:
Closing as the `iCubGenova04` reference here has been transmogrified to `iCubNancy01`, and it is a different robot to the `iCubGenova04` as of November 2016.

cc @julijenv

closed 2016-11-17 by traversaro