robotology/mesh-support #124: HW fault for quadrature encoder

francesco-romano, 2015-11-26:
We have recently seen the following error which causes the joint to got into hardware fault

```
[ERROR] from BOARD 10.0.1.6 (leftLeg), src CAN1, adr 4, time 736s 250m 523u: (code 0x0200000d, par16 0x0003 par64 0x00000000000014a5) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + INFO = NO MORE
```

Putting the joint in idle and run it again seems to fix the problem

ale-git, 2015-11-26:
I suggest to check left upperleg motor encoder cables in 2FOC board with can address 4. 
Since the error is generated by a cross check with hall sensors, a false positive could be generated by a defective hall sensor (in some robots there was a problem of missing spacers in the rotor).

super-ste, 2015-11-26:
can we please add the reference to the robot as Label?

super-ste, 2015-11-27:
@julijenv can you have a look at this today?

francesco-romano, 2015-12-10:
This happened also on the right upper arm:
```
[ERROR] from BOARD 10.0.1.3 (rightUpperArm), src CAN1, adr 1, time 189s 181m 541u: (code 0x0200000d, par16 0x0000 par64 0x0000000000001c20) -> MC: 2FOC quadrature encoder phase broken. The motor encoder is not counting even if the motor is moving, please check motor encoder cables. In par64 0xFF is the mask of raw encoder value. par16 = ID of joint. + .
```

francesco-romano, 2015-12-14:
It would be very (very) useful if the diagnostic could discern the following two cases:

- no reading of the encoder (so issues with the cable)
- wrong count of the ticks

This can help determining the correct maintenance

@ale-git what do you think?

cc @maggia80

ale-git, 2015-12-14:
@francesco-romano at present, there are already two different messages for "no reading" and "optical disk dirty", but the second error can be detected only in a motor encoder with Index signal (i.e. iCub V3 motors). Thus, an encoder error in a V2 robot is always a "no reading", since an absolute precise reference is needed to detect dirty optical disks. With iCub V3, we'll be able to detect also wrong counts.

francesco-romano, 2015-12-15:
This morning @randaz81 and me run two tests on the robot.

- Encoder consistency test (check between motor and joint size encoders)
- Encoder drift test

Both were successful (in the sense that the run correctly). At this point I think we can exclude the presence of dirt on the encoder. A cable contact seems to be the most plausible cause.

cc @super-ste @julijenv

julijenv, 2015-12-16:
It has been unmounted the backpack and recrimp all the wires from the optical sensor, we relaunch the robot and look like that is better, let keep an eye on this.

julijenv, 2015-12-18:
can we close it?
@francesco-romano @DanielePucci

francesco-romano, 2015-12-18:
Closing the issue for now...

closed 2015-12-18 by francesco-romano