robotology/mesh-support #125: J10 and J11 of the arm moving autonomously

robertocalandra, 2015-11-26:
I am getting a strange behavior when controlling the first 5 joint of the arms of the robot using position control trough a C code.
Somehow, J10 (thumb_distal) and J11 (index_proximal) decide to flex autonomously.

I don't know if it depends from the way I call the position control interface (below) or from something else. Any idea?

```
Vector command;
int nj=0;
Rpos->getAxes(&nj);
command.resize(nj);
for (int i=0;i<5;i++){
    command[i] = initPos[i];
}
Rpos->positionMove(command.data());
```

barbalberto, 2015-11-26:
Yes, the  ``positionMove(double *)`` is meant to send a reference for **all** the joints, so the receiver simply reads out of the memory untill it gets one reference for each joint.
If you want to control a subset of joints you have to use the so called "group joint" version
``virtual bool positionMove(const int n_joint, const int *joints, const double *refs);``

described here :https://github.com/robotology/yarp/blob/master/src/libYARP_dev/include/yarp/dev/IPositionControl2.h@153

pattacini, 2015-11-26:
@robertocalandra next time, could you please post software related issues relying on proper tracking systems, as per our [**guidelines**](https://github.com/robotology/icub-support/blob/master/CONTRIBUTING.md)?

For your convenience, they will show up any time you click on the **New Issue** button.

robertocalandra, 2015-11-27:
@barbalberto Thank you! (the link is broken)

closed 2015-11-27 by robertocalandra

pattacini, 2015-11-27:
Here is the doc for [**`IPositionControl2`**](http://www.yarp.it/classyarp_1_1dev_1_1IPositionControl2.html).

In particular, look for the following method:
```cpp
positionMove(const int n_joint, const int *joints, const double *refs);
```