robotology/mesh-support #130: Report of issues encountered during IGT demo (2 December 2015)

francesco-romano, 2015-12-03:
### Misc

- Left shoulder tendons continuously break
- Warning on "timing overflow on some phases of the control Loop on EMS" (arms)
- Errors (and the whole part goes into HW fault) on quadrature encoders (right shoulder, left knee)
- Check cover and joint limits (cover hits the waist motor and we risk to cut the motor cables)
- Try to find an alternative to icub-srv because we do not scale in case of problem on it. Alternatively we should add a small VGA monitor to the setup.

### Breathers & funny-things related

- Sometime the arm(s) drifts and the only possible way to stop it is to close the breathers and issue an home command. Once the home command has been even ignored.
- Shaking of the arms. Maybe this is related to the warning on slow control loop we saw on the robotInterface log
- Check breathers on/off functionality  

cc @davege89 @super-ste @marcoaccame @DanielePucci @julijenv 
cc @lornat75 @iron76

spada1978, 2015-12-04:
proximal right thumb repaired

francesco-romano, 2015-12-10:
Added reference to single issues:

https://github.com/robotology/icub-support/issues/124
https://github.com/robotology/funny-things/issues/8
https://github.com/robotology/funny-things/issues/9

francesco-romano, 2016-01-29:
Closing (this was only a reminder)

closed 2016-01-29 by francesco-romano