robotology/mesh-support #131: Tendon number 8 broken

DanielePucci, 2015-12-10:
While doing a YOGA demo, the tendon broke. 

Thanks @julijenv who is already working on that.

super-ste, 2015-12-10:
Guys, guys
before doing any more YOGA demos or waving arm around in an uncontrolled fashion, let's make sure we troubleshoot and solve the issue #64  

cc @julijenv 
@DanielePucci 
@francesco-romano

DanielePucci, 2015-12-10:
The point is that we were performing the yoga demo in torque control, so this rules out an eventual PID related problem causing the issue. Probably, the left shoulder of iCubGenova02 requires an inspection from a mechanical perspective for ruling out an hardware problem.

francesco-romano, 2016-01-29:
Let's close this issue for now.
The tendon has already been replaced.

closed 2016-01-29 by francesco-romano