robotology/mesh-support #132: finger middle proximal not moving

fdarold, 2015-12-11:
Hi everyone,

We are experiencing a strange problem with our iCub @ Plymouth. 

During calibration the middle proximal of the left arm works fine, but after that it is not responding to our commands. 

If we try to physically interact with the arm moving that joint the motor rotates as it is not stiff. 

Thanks for  advices and help.

fdarold, 2015-12-13:
Sorry for posting again, middle proximal is not activated during calibration. 

Apologies!

Basically the motor is idle all the time, but the robot motor GUI sees it in position control.

Thanks and sorry for double posting.

julijenv, 2015-12-15:
Hi @fdarold ,
before doing any of the following suggestion please shut down the motors switch.

there are 4 explanations for your possible problem:

1) Hardware - mechanical play - at the level of the proximal cable, please try to open and close the proximal joint by spinning clockwise and counterclockwise the screw of the shaft of the motor. (see picture below)
[image]

Do you have play (means that there is no direct movement of the middle finger for a bit of movement of the screwdriver) or worse case scenario no closing or opening action in any direction you spin the screw?

In the first case, you will need to tense the cable of the proximal joint : 
 For V1 forearm, you will need to open the cover of the forearm and follow the cable which go out of the pulley - you should see a tensionner, little piece of alu minium with a perforated screw allowing you to increase the disctance of the trace of the wire by unscrewing the screw out of the tensionner until getting tense wire
For V2 version you will need to take a look at the first phalange of your issued finger and see beneath teh transparent plastic holding the aside wires ofthe finger. Below you will find a screw nut holding the just pull out the knot of the end of the wire until getting the cable tense and close the nut afterwards to keeep the tensionning.

[image]

In the second case you will need to rewire the proximal joint. Let us know

2) Hardwarre - chipset - above the MAIS board above the knuck of the hand. It might be a misalignement of fhte FC boards (5 little rectngular boards on the top of the longest borad above the knuck)

3) Hardware - motor wires broken - you will need to open the upparm cover and it might be tricly please contact us before doing it.

for any direct help get in touch on skype ID : icub-support

closed 2018-01-16 by julijenv