robotology/mesh-support #133: iCub calibration

tonymorse, 2015-12-11:
The Plymouth iCub is not calibrating correctly, joint 0 on both the left and right are not moving to the correct position. If we use the yarpmotorgui to move the arms to the fine calibration position, we can't move joint 0 on either arm far enough. when we set it to -95 the upper arm is still angled down away from the body. I would guess it's out by about 30 degrees (same amount on both arms).

not sure if we can simply change the limits to enable setting a fine calibration point or if it's more complex than that???

tonymorse, 2015-12-11:
just tried doing the fine calibration and it hasn't changed anything

tonymorse, 2015-12-11:
To clarify, the number in the file are now different but the arm still calibrates to the wrong position and cannot be raised any further than before.

tonymorse, 2015-12-11:
might be a silly question but how does the iCubInterface know this is the plymouth robot and therefore load the correct configuration scripts?? (just thinking as changing the config file made no difference)

traversaro, 2015-12-11:
You may find details on how the `robotInterface` find the correct files (using the `YARP_ROBOT_NAME` enviromental variable) at http://wiki.icub.org/wiki/Configuration_files .

barbalberto, 2015-12-14:
which files are you changing? I mean in which folder? When robotInterface starts, it prints out the files it is reading from. Do they match with the files you are updating?

closed 2018-01-16 by julijenv