robotology/mesh-support #139: Left index not closing on startup

tanismar, 2015-12-16:
Hi,  after the left arm has been repaired and reattached, I noticed that the left index was not closing during the calibration stage. I don't really need it too much, but maybe some people do afterwards, so It 'll be nice to get it repaired.

Thanks!

vtikha, 2015-12-16:
@tanismar please specify which robot you are talking about. Add the tag of the robot name to the issue, and put the name also in the title.

pattacini, 2015-12-16:
not the robot name in the title :smirk: 
see [**`guidelines`**](https://github.com/robotology/icub-support/blob/master/CONTRIBUTING.md).

vtikha, 2015-12-16:
I still put it there to please @super-ste :smile:

alecive, 2015-12-16:
... and me ! :dancer:

pattacini, 2015-12-16:
I can always :gun: and :bomb: you, but it's :christmas_tree: :santa:

super-ste, 2015-12-16:
Me thinks okay what  @pattacini  does.
Me thinks stinky :us: should be kept out this repo.

AUGH
[image]

julijenv, 2015-12-18:
Has been repaired the day before yesterday by @spada1978

closed 2015-12-18 by julijenv