robotology/mesh-support #145: Default password for old pc104 images

rlober, 2016-01-06:
I am testing the image `image4.0-dhcp.img` for pc104 on iCubParis01 and I can't find the password anywhere. I looked here: http://wiki.icub.org/wiki/The_Linux_on_the_pc104_oldrepo , but all it says is 

>**Important:** the images come with a dummy password for icub and root users, we recommend you keep the pc104 disconnected form the external network or modify the password.

I have tried the following:
- icub
- icubicub
- iCubAdmin
- admin
- pass
- password
- and others...

It would be cool to list all of the default passwords for the various images.

kt10aan, 2016-01-07:
Have you tried "live" as the password (without the quotes)?
Also, the page you are reffering to, has instructions for images < v4.
This is the correct documentation: http://wiki.icub.org/wiki/The_Linux_on_the_pc104

rlober, 2016-01-07:
Hello @kt10aan!

Yes I tried `live` as a password. 

If it is written in the wiki anywhere I have tried it probably.

barbalberto, 2016-01-07:
Password is not publicly written because that was the policy at that time, anyway it is suggested to go for the latest version of image, which now is the 5.0 you can find here http://wiki.icub.org/wiki/The_Linux_on_the_pc104.

Do you have any particular need to stick with the version 4?

rlober, 2016-01-07:
Yes. The latest version was not booting on pc104 of iCubParis01. I tried burning the image twice, once with the shell script you guys wrote, and once via `dd`. Neither method produced a usb bootable from our robot.

barbalberto, 2016-01-07:
Ok, I see. I've updated the wiki page for version 4 with the correct password.
it is 'icubAdmin'  (lowercase icub)

closed 2016-01-07 by barbalberto

rlober, 2016-01-07:
:+1: @barbalberto 

Thanks a bunch man!