robotology/mesh-support #149: [robotInterface] Interesting sensor failures with new firmware and icub-main

drdanz, 2016-01-08:
[image] **Issue by [rlober](https://github.com/rlober)**
_Thursday Jan 07, 2016 at 16:26 GMT_
_Originally opened as https://github.com/robotology/icub-main/issues/271_

----

Not sure if this is exactly the best place for this issue but I figured because it happens in `robotInterface` why not.

Anyways, I recently updated iCubParis02 to the newest firmware and the latest `yarp` and `icub-main` and I am getting some interesting error messages from `robotInterface`. I say interesting because I have never seen them before and I am not sure where to start resolving them. After speaking with @julijenv it seems there have been some big changes to the firmware before the holidays and I am wondering if we are seeing some new bugs. Anyways here are some examples of these errors:

```
3015		32,457671		ERROR		AxisName not found
3493		36,865772		ERROR		Trying to get fullscale data from sensor 14 net [1]: no answer received or message lost (ch:0)
5206		212,515410		ERROR		AnalogWrapper: left_leg: Sensor returned timeout error (code 3).
```
*edit: these are only some of the errors...*

Here is the full log: [log_pc104_robotInterface_5489.txt](https://github.com/robotology/icub-main/files/81375/log_pc104_robotInterface_5489.txt)

Addmittedly these *could* be hardware related, but I am not entirely sure where to start with the debugging. 

For reference here is the `CMakeCache` of `icub-main` for pc104: [CMakeCache_icub-main-build-pc104.txt](https://github.com/robotology/icub-main/files/81368/CMakeCache_icub-main-build-pc104.txt)

cc: @randaz81 

edit:
Also. lots of warnings about syntax errors in the xml files 

```
2592		30,562498		WARNING		Invalid syntax while loading /usr/local/src/robot/source/icub-main/build-pc104/share/iCub/robots/iCubParis02/./icub_all.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 

```

drdanz, 2016-01-08:
[image] **Comment by [Tobias-Fischer](https://github.com/Tobias-Fischer)**
_Thursday Jan 07, 2016 at 16:29 GMT_

----

See e.g. https://github.com/robotology/icub-main/commit/8c0c9ae93a77565247adca016ef9f9138578fc5c for the `AxisName` problem.

No clue about the other stuff ;)

drdanz, 2016-01-08:
[image] **Comment by [Tobias-Fischer](https://github.com/Tobias-Fischer)**
_Thursday Jan 07, 2016 at 16:30 GMT_

----

Ahh, for the warnings, add `<!DOCTYPE devices PUBLIC "-//YARP//DTD robotInterface 1.0//EN" "http://www.icub.org/DTD/robotInterfaceV1.0.dtd">` to the XML files. Again, https://github.com/robotology/icub-main/commit/8c0c9ae93a77565247adca016ef9f9138578fc5c shows an example.

drdanz, 2016-01-08:
[image] **Comment by [rlober](https://github.com/rlober)**
_Thursday Jan 07, 2016 at 16:33 GMT_

----

@Tobias-Fischer 

Thanks for the info! I will make those changes.

drdanz, 2016-01-08:
[image] **Comment by [rlober](https://github.com/rlober)**
_Friday Jan 08, 2016 at 12:27 GMT_

----

**Updates: New error**
```
5130 311,814836 5 recv buffer overrun ( 500 >  500) 
```
After running for about 5 min, `robotInterface` completely crashes pc104 and I get this `buffer overrun` error. 

Ah yes... Here is the log: 
[log_pc104_robotInterface_3397.txt](https://github.com/robotology/icub-main/files/82672/log_pc104_robotInterface_3397.txt)

rlober, 2016-01-12:
@randaz81 

Ran the test this morning as asked. Here is the log file:
[log_pc104_robotInterface_10857.txt](https://github.com/robotology/icub-support/files/87018/log_pc104_robotInterface_10857.txt)

randaz81, 2016-01-12:
Unfortunately, even with the additional debug message it's impossibile to understand what is the source of the CAN buffer overrun. I'd proceed as follows: 
edit your icub_all.xml (double check that you are editing the copy that is effectively used by robotInterface) and remove, one by one, the FT sensor block, the skin block etc. Every time you restart robotInterface you have to resert (turn off) the control boards. If the error disappears removing one part, we'll focus on that specific part. I also recommend a manual update of the firmware of FT sensors and MAIS.

rlober, 2016-01-12:
Ok, I will do the following:

 -  update the firmware manually
 -  test without the FT sensor block
 -  test without the skin block
 -  figure out which causes the error
 -  test each individual line till the error is localized

rlober, 2016-01-12:
@randaz81 @julijenv 

We updated all of the STRAIN sensor firmware to the newest version: `3.0.4`. We then ran `robotInterface` and after about 3 minutes got the following errors:
```
20065		14579,375883		ERROR		run()-pushReadMsg() failed on CAN bus 3
20066		14579,759043		ERROR		recv buffer overrun ( 500 >  500)
20068		14579,981051		ERROR		run()-pushReadMsg() failed on CAN bus 4
20070		14579,981160		ERROR		AnalogWrapper: right_leg: Sensor returned timeout error (code 3).
20073		14579,981229		ERROR		recv buffer overrun ( 500 >  500)
```
`robotInterface` then killed itself. Note that the errors repeated many times. This is just a sample.

We then commented out all left and right leg FT sensor lines in `icub_all.xml`:
```xml
<!-- VIRTUAL ANALOG SERVERs -->
    <devices file="wrappers/VFT/left_arm_VFT_wrapper.xml" />
    <!-- <devices file="wrappers/VFT/left_leg_VFT_wrapper.xml" /> -->
    <devices file="wrappers/VFT/right_arm_VFT_wrapper.xml" />
    <!-- <devices file="wrappers/VFT/right_leg_VFT_wrapper.xml" /> -->
    <devices file="wrappers/VFT/torso_VFT_wrapper.xml" />

    <devices file="hardware/VFT/left_arm_virtual_strain.xml" />
    <!-- <devices file="hardware/VFT/left_leg_virtual_strain.xml" /> -->
    <devices file="hardware/VFT/right_arm_virtual_strain.xml" />
    <!-- <devices file="hardware/VFT/right_leg_virtual_strain.xml" /> -->
    <devices file="hardware/VFT/torso_virtual_strain.xml" />

    <!-- REAL ANALOG SENSORS -->
    <devices file="wrappers/FT/left_arm_FT_wrapper.xml" />
    <!-- <devices file="wrappers/FT/left_leg_FT_wrapper.xml" /> -->
    <devices file="wrappers/FT/right_arm_FT_wrapper.xml" />
    <!-- <devices file="wrappers/FT/right_leg_FT_wrapper.xml" /> -->
    <devices file="wrappers/MAIS/left_hand_mais_wrapper.xml" />
    <devices file="wrappers/MAIS/right_hand_mais_wrapper.xml" />
    <devices file="hardware/FT/left_arm_strain.xml" />
    <!-- <devices file="hardware/FT/left_leg_strain.xml" /> -->
    <devices file="hardware/FT/right_arm_strain.xml" />
    <!-- <devices file="hardware/FT/right_leg_strain.xml" /> -->
    <devices file="hardware/MAIS/left_hand_mais.xml" />
    <devices file="hardware/MAIS/right_hand_mais.xml" />

    <devices file="wrappers/FT/left_foot_FT_wrapper.xml" />
    <devices file="wrappers/FT/right_foot_FT_wrapper.xml" />
    <devices file="hardware/FT/left_foot_strain.xml" />
    <devices file="hardware/FT/right_foot_strain.xml" />
```
and re-ran `robotInterface`. After a longer period of time we got the following errors:
```
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [4] printing boards infos:
None
[DEBUG]sharedcan [0] thread ran 595 times, req.dT:10[ms], av.dT:10.08[ms] av.loopT :0.02[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [0] printing boards infos:
None
[DEBUG]sharedcan [0] thread ran 595 times, req.dT:10[ms], av.dT:10.08[ms] av.loopT :0.01[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [0] printing boards infos:
None
[INFO]icub/cartesianController/left_arm: Checking if cartesian solver cartesianSolver/left_arm is alive... not yet
[INFO]icub/cartesianController/right_arm: Checking if cartesian solver cartesianSolver/right_arm is alive... not yet
[[[ERROR]recv buffer overrun ( 500 >  500)
ERROR]recv buffer overrun ( 500 >  500)
ERROR]recv buffer overrun ( 500 >  500)
[[ERROR]AnalogWrapper: left_arm: Sensor returned timeout error (code 3).ERROR]AnalogWrapper: right_arm: Sensor returned timeout error (code 3).

[[[[[[ERROR]recv buffer overrun (2047 > 2047)ERRORERROR]recv buffer overrun ( 500 >  500)
ERROR]recv buffer overrun (2047 > 2047)
]recv buffer overrun ( 500 >  500)
ERROR
[... 50 lines cut ...]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [3] printing boards infos:
None
[DEBUG]sharedcan [4] thread ran 594 times, req.dT:10[ms], av.dT:10.09[ms] av.loopT :0.02[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [4] printing boards infos:
None
[DEBUG]sharedcan [0] thread ran 594 times, req.dT:10[ms], av.dT:10.09[ms] av.loopT :0.02[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [0] printing boards infos:
None
[DEBUG]sharedcan [0] thread ran 594 times, req.dT:10[ms], av.dT:10.09[ms] av.loopT :0.01[ms]
[DEBUG]     Device has no ICanBusErr interface
[DEBUG]sharedcan [0] printing boards infos:
None
[DEBUG]sharedcan [1] thread ran 583 times, req.dT:10[ms], av.dT:10.21[ms] av.loopT :0.21[ms]
Killed
icub@pc104:~$ 
```

Here are the `yarp logs` in order:

[log_pc104_robotInterface_3965.txt](https://github.com/robotology/icub-support/files/87463/log_pc104_robotInterface_3965.txt)
[log_pc104_robotInterface_4160.txt](https://github.com/robotology/icub-support/files/87464/log_pc104_robotInterface_4160.txt)
[log_pc104_robotInterface_4541.txt](https://github.com/robotology/icub-support/files/87465/log_pc104_robotInterface_4541.txt)

randaz81, 2016-01-13:
The "killed" message was due to a wrong commit in yarp causing a memory leak and fixed the same day by d3846669c1f32fa2a11285a513d85722d11b0b89.
The other issues are still under investigation.

rlober, 2016-01-14:
Ok, I have been playing with the robot for a couple of days now and I think these issues were fixed with the yarp bug described here: https://github.com/robotology/QA/issues/86

Thanks for the help @randaz81 @julijenv

closed 2016-01-14 by rlober