robotology/mesh-support #152: Not able to boot on iCub Live 5 (15.11.02) PC104

ahoarau, 2016-01-12:
Hello,
I'm having trouble booting on *icub-live_5.0-15.11.02-rt-amd64.iso* for pc104.
I tried : 
- The script on Ubuntu 14.04 -> fail
- The script on Debian 7 -> some warnings, but can't boost
- USB Creator on Ubuntu -> Invalid string GNU/Linux
- sudo dd if=<image> of=/dev/sdb bs=4M -> can't boot
- unetbootin -> no error but can't seem to boot either
I loaded the image on a virtualbox (as linux 64) and it works.. What to do next ? Thanks !

mbrunettini, 2016-01-13:
Hi,
if you having problem with the  icub_iso2usb.sh (it is not specified here), please check that:
(as described http://wiki.icub.org/wiki/The_Linux_on_the_pc104#How_to_use_the_icub_iso2usb.sh_script )
1) you are running it on a supported linux, the script is only compatible with Debian GNU/LInux 7 [wheezy] and Ubuntu Ubuntu 14.04.3 LTS [Trusty]
2) the script has some required packages, please check that they are installed on your system.

I think the problem is with the missing mkdiskimage command, wich is included in the package syslinux-common (on debian 7), but I can't be sure without any kind of log.

In case the USB image is still not working, in order to check the problem we need:
1) which system (with version) you are using to create the USB
2) the log of ALL errors/warning produced by the script.

Let us know.
Bye

mbrunettini, 2016-01-13:
More in the details:
>The script on Ubuntu 14.04 -> fail
The script works on Ubuntu 14.04.03. If it fails in your system, It can be useful to report the errors

>The script on Debian 7 -> some warnings, but can't boost
Again, it is useful to report logs.

>USB Creator on Ubuntu -> Invalid string GNU/Linux
This system is not supported, please follow the documentation.

>sudo dd if= of=/dev/sdb bs=4M -> can't boot
As above.

>unetbootin -> no error but can't seem to boot either
As above.

ahoarau, 2016-01-13:
Tried again this morning on Ubuntu, the script worked :/ on Ubuntu 14.04.3. 
But still not able to boot on pc104 (I'm trying on iCubParis01).
```
checking mkfs.ext4 executable..
/sbin/mkfs.ext4
checking install-mbr executable..
/sbin/install-mbr
checking syslinux executable..
/usr/bin/syslinux
checking 7z executable..
/usr/bin/7z
checking mkdosfs executable..
/sbin/mkdosfs
checking mkdiskimage executable..
/usr/bin/mkdiskimage
[13-01-2016 10:24:53] ./icub_iso2usb.sh started: please do not unplug device sdb until finished.
Warning: You requested a partition from 744MB to 6858MB.                  
The closest location we can manage is 780MB to 6858MB.
Is this still acceptable to you?
Yes/No? Yes                                                               
Information: You may need to update /etc/fstab.                           

umount2: Invalid argument
umount: /tmp/tmp.6Z7yrqH8rv: not mounted
[13-01-2016 10:31:14] ./icub_iso2usb.sh ended successfully: you can now unplug device sdb.
```

mbrunettini, 2016-01-13:
And what about the command line parameters you used with the script?
Do you issued it with as superuser?

ahoarau, 2016-01-13:
```
sudo ./icub_iso2usb.sh -f icub-live_5.0-15.11.02-rt-amd64.iso -t sdb
```

mbrunettini, 2016-01-13:
Everything seems to be correct.
Are you sure it does not boot? Or maybe it fails in some other step, i.e. mounting remote NFS resources? 
Which is the problem whith your boot?

ahoarau, 2016-01-13:
How can I be sure it's booted other than seeing the computer on the network (i use arp-scan, which shows 10.0.0.2 if I put the image4.0) ?

mbrunettini, 2016-01-13:
By connecting a monitor to the PC104

ahoarau, 2016-01-13:
TIL we can connect a monitor to the pc105 :v

mbrunettini, 2016-01-13:
@julijenv : can you help Antoine in connecting a monitor to the PC104 and check what's wrong?

super-ste, 2016-01-13:
@ahoarau 
Give us a couple of days. We'll craft, test and validate a super-easy step by step procedure on how to hack an old VGA cable to add a monitor to iCub PC104

ahoarau, 2016-01-13:
Perfect ! Keep me posted !

super-ste, 2016-01-18:
Hi @ahoarau,
@julijenv nicely put the information together in this doc:

https://svn.robotology.eu/repos/iCubHardware-pub/trunk/mechanics/public/doc/howTo/Connect_VGA_to_PC104.pdf

(which is also linked from the iCub Wiki) you should be able to savage a spare VGA cable you have hanging around and crimp/solder the wires directly to a 2.54mm pin header. [fastest way]

else
we can ship you one we manufactured already [might take a little longer]

Let us know what you prefer.

super-ste, 2016-01-21:
Hi @ahoarau ,
are you working on iCubParis01 or iCubParis02 (so we can add the proper label and correctly track this  issue)

ahoarau, 2016-01-22:
I haven't had time yet to work on this yet, but thanks for the tuto, I'll keep you posted.
It is indeed Paris01.

ahoarau, 2016-05-31:
It has been "fixed" thanks to @jeljaik ! The partition might have been the issue all along.

closed 2016-05-31 by ahoarau