robotology/mesh-support #156: Right arm 4DC and/or STRAIN problems

gsaponaro, 2016-01-13:
Hi, we have some problems in the right_arm CAN network of our robot (network 2). We verified all the relevant connections and wires. Here is our analysis so far, which likely isolates the problem in the 4DC control board (see third experiment below) and perhaps also in the STRAIN force/torque sensor.

1. With all the boards of this CAN network connected, **canLoader** shows:
[image]

2. We disconnected the BLLs of this CAN network (one at a time, first the bottom then the top), and in both cases **canLoader** again showed problems:
[image]
[image]
therefore the two BLL boards are fine.

3. We disconnected the 4DC board and obtained no errors in **canLoader** -> 4DC faulty.

4. We disconnected the STRAIN f/t sensor, no errors in **canLoader**, but errors when starting the **robotInterface** calibration procedure. Full log is [here](https://gist.github.com/gsaponaro/da8114d4207f5ca494d0), excerpt of relevant parts and **yarpmotorgui** screenshot below:
```
[ERROR]sharedcan [2] thread:1 msg:7 joint:4 timed out
[ERROR]readWord16: message timed out
...
[ERROR]100ms Timeout occured in setControlModeRaw ( j:4), current mode: cal, requested: pos
...
[ERROR]100ms Timeout occured in setControlModeRaw ( j:6), current mode: cal, requested: pos
...
[ERROR]Right_Arm_Calibrator : checkGoneToZeroThreshold: joint  4  Timeout while going to zero! 
```

[image]

Our thoughts so far:
* we probably need a replacement 4DC;
* question: could the **canLoader** problems also be caused by the STRAIN f/t sensor? Any other suggestions?

/cc @lorejam

julijenv, 2016-01-15:
Hi @gsaponaro,
As more or less anticipated yesterday, I guess that as the BLL number ID 2 is not visible, so it means that something weird happen !!!
FIRST:
I'd like you to:
-> Do a power cycle of the motors. (switch OFF (wait 10s) and ON - or just TURN ON if u just restart the robot)
-> disconnect the BLL and MC4 (4DC) and do canLoder.
make a screenshot
SWITCH OFF the motors
-> disconnect the FT SENSOR and reconnect the MC4 (4DC) and do canLoder.
make a screenshot
By doning so we will nkow which boardd is actually making this UNKNWON Board appearing.
Once finding the responsible.

SECOND:

I would highly suggest u to reflash the boards (BLL and 4DC - !!! BUT NOT THE FT sensor!!!, ticking the EEPROM case on the right side inside canLoader software) on the right arm.
Do that in 2 steps:
Do a power cycle of the motors. (switch OFF (wait 10s) and ON - or just TURN ON if u just restart the robot)
wait 5s
select the firmware required by the board, tick the EEPROM and download the firmware.
repeat that for all the boards (BLL and 4DC)

THIRD
SHUT DOWN THE MOTORS
replug all the filters to all their corresponding boards and check with canLoader
if you still have this UNKNOWN message on.
Let us know

gsaponaro, 2016-01-15:
Thank you, just a clarification about the first test: when you say "disconnect the BLL", you mean disconnect both the BLLs on net 2 (2B0 and 2B1), correct?

julijenv, 2016-01-15:
well actually id like you to make a test on every single board so just let connected 1 board at the time.
is it clearer like that?

gsaponaro, 2016-01-15:
OK. In the following tests (and also the ones from yesterday), we always do a power cycle and verify consistency of results by repeating the same test ~5 times. We have also replaced all the CAN filters.

As you can see below, if we leave only 1 or 2 boards connected at a time, there are no problems.  Problems only show up when 3+ boards are connected.

**First**

Having disconnected bottom BLL and the MC4 (4DC), thus having left only top BLL and STRAIN:

[image]

Having disconnected top BLL and the MC4 (4DC), thus having left only bottom BLL and STRAIN:

[image]

Having disconnected both BLLs and the MC4 (4DC), thus having left only STRAIN:

[image]

Having disconnected everything except MC4 (4DC):

[image]

**Second**

We reflashed the 4DC with EEPROM option successfully (4DC.1.19.out.S, build 85).

We reflashed the BLL with id 1 with EEPROM option successfully (2BLL.1.50.out.S, build 85).

We reflashed the BLL with id 2 with EEPROM option successfully (2BLL.1.57.out.S, build 85).

**Third**

Reconnected everything, power flashed as usual, waited, and... the problem persists:

[image]

**Our opinion so far**

The BLLs are fine (see also experiment 2 of our first message in this issue from yesterday, where one BLL at a time was connected and there were 3+ boards -> always problems).
The culprit seems to be 4DC (most likely) or STRAIN, because they are the ones present in the error situations.

kt10aan, 2016-01-16:
@gsaponaro, we had some issues with board recognition too. The usual debugging and trouble shooting procedure  was not helping and the problem was at the filter. See https://github.com/robotology/icub-support/issues/91.

Also, as @julijenv mentioned, sometimes the flashing of the boards has to be done again, manually. See also https://github.com/robotology/icub-support/issues/144

gsaponaro, 2016-01-19:
Hi @kt10aan, thanks for the input. We have already verified both of those things (CAN filters, and manual re-flashing of the firmware, except STRAIN as per @julijenv's last message).

@julijenv: we have performed the tests you requested (see a couple of messages up), waiting for the next suggestion when you can, thanks.

*if we leave only 1 or 2 boards connected at a time, there are no problems. Problems only show up when 3+ boards are connected.*

julijenv, 2016-01-19:
HI @gsaponaro, did you try to plug everything but the STRAIN?
This could definitely helping us finding if it is the STRAIN is the responsible for this mess!
In any csae we are arranging things to send you a spare MC4.
How does it sounds?

gsaponaro, 2016-01-19:
Hi there,

Yes, we made that test too - screenshot above, where it says *Having disconnected both BLLs and the MC4 (4DC), thus having left only STRAIN* -> **canLoader** showed STRAIN ok, consistently and after power cycles.

Yes, having the spare MC4 would be great and helpful. :) Let's continue the discussion by email about the delivery then.

I'll update this issue once we make more tests with the spare.

julijenv, 2016-01-19:
@gsaponaro , no Giovanni Im asking you to plug everything EXCEPT (BUT) the STRAIN, which means MC4 and both BLLs. :P

gsaponaro, 2016-01-19:
Oh sorry, reading too fast here :)

Yes, in fact we did that test too (Experiment 4 in the very first message of this issue) -> it gave calibration errors, full **robotInterface** log attached in that message ([direct link](https://gist.github.com/gsaponaro/da8114d4207f5ca494d0)).

julijenv, 2016-01-19:
@gsaponaro ,
Could you just make a canLoader test; that's it nothing more and see if he bad message show up.
VBecause of course not having the strain brings the robot to do not calibrate, except maybe if your remove all the sentence in icub_all.xml containing the FT sensro for your damaged arm.
Just do canLoader

gsaponaro, 2016-01-19:
Sure, will do it first thing tomorrow.

(In the previous test, indeed we had commented out the FT parts from ```icub_all.xml```.)

gsaponaro, 2016-01-20:
The situation when STRAIN is unplugged while BLLs + MC4 are plugged is consistently like this (we tried it many times including power cycles):

1.
**canLoader** shows fine:
[image]

2.
However, when there are more packets and traffic on this CAN network (i.e., calibration procedure with properly modified xml, commenting STRAIN F/T parts), we get errors:
```
[ERROR]sharedcan [2] board 4 FAULT OVERLOAD AXIS 0
...
[ERROR]sharedcan [2] board 3 FAULT OVERLOAD AXIS 1
```

Delivery address and contacts have been sent by email yesterday night, thanks.

gsaponaro, 2016-01-22:
Good news: replacement MC4 installed, arm works smoothly after trying **canLoader**, **robotInterface** and moving all joints with **yarpmotorgui**. Many thanks. :)

Just a small issue left: occasionally (20% of the times), **canLoader** is showing some junk in the "Additional info" string of a BLL or STRAIN board, like in the screenshot below. Even after a clean power cycle. Any thought about this?
 
[image]

gsaponaro, 2016-01-26:
Funny... the junk characters are not appearing anymore now. If they do, we'll re-flash the corresponding firmware with EEPROM option selected, as suggested on Skype. Thanks again, issue solved.

closed 2016-01-26 by gsaponaro