robotology/mesh-support #158: Problem finding INI files, robotInterface won't start

marijnfs, 2016-01-15:
The robotInterface won't start on our robot (Lugano iCub). It complains about .ini files that can't be found, even though it seems it the environment variables point to it:

    [ERROR]Cannot find "cartesiancontrollerserver" plugin (not built in, and no .ini file found for it)
    [ERROR]Check that YARP_DATA_DIRS leads to at least one directory with plugins/cartesiancontrollerserver.ini or share/yarp/plugins/cartesiancontrollerserver.ini in it
    [ERROR]Cannot find "sharedcan" plugin (not built in, and no .ini file found for it)
    [ERROR]Check that YARP_DATA_DIRS leads to at least one directory with plugins/sharedcan.ini or share/yarp/plugins/sharedcan.ini in i

The files are there:
    icub@pc104:~$ locate plugins/sharedcan.ini
    /usr/local/src/robot/icub-main/build-pc104/share/iCub/plugins/sharedcan.ini

    icub@pc104:~$ echo $YARP_DATA_DIRS 
    /usr/local/src/robot/yarp/build-pc104/share/yarp:/usr/local/src/robot/icub-main/build-pc104/share/iCub:/usr/local/src/robot/iCubContrib/share/ICUBcontrib

pattacini, 2016-01-15:
To check which plugin is currently recognized you can do:
```bash
yarp plugin --list
```

To check whether the _resource finding_ mechanism based on `YARP_DATA_DIRS` is actually working you can do for example on the `pc104`:

```bash
yarp resource --context robotInterface --find robotInterface.ini
```
Is the `robotInterface.ini` file found?

Lately I've seen a bunch of issues being discussed on the interaction between `YARP_DATA_DIRS` and the `path.d` mechanisms: @drdanz and @traversaro might shed a light on this.

barbalberto, 2016-01-15:
In  addition, ca you also verify that those modules `cartesiancontrollerserver` and 'sharedcan' are selected to be compiled in the iCub's CMake flags?

marijnfs, 2016-01-15:
I have selected those modules now.
It is mainly complaining about *canBusAnalogSensor.ini*

Check that YARP_DATA_DIRS leads to at least one directory with plugins/canBusAnalogSensor.ini or share/yarp/plugins/canBusAnalogSensor.ini in it

I can find some of the files in */usr/local/src/robot/icub-main/src/libraries/icubmod/canBusAnalogSensor/canBusAnalogSensor.ini*

This *icubmod* directory is not in the env variables, is this a new place for the files perhaps?.

The resource seems found:

icub@pc104:~$ yarp resource --context robotInterface --find robotInterface.ini
||| configuring
||| no policy found
||| added context robotInterface
||| finding file [robotInterface.ini]
||| checking [/home/icub/robotInterface.ini] (pwd)
||| checking [/home/icub/.config/yarp/robots/iCubLugano01] (robot YARP_CONFIG_HOME)
||| checking [/home/icub/.local/share/yarp/robots/iCubLugano01] (robot YARP_DATA_HOME)
||| checking [/etc/yarp/robots/iCubLugano01] (robot YARP_CONFIG_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/robots/iCubLugano01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubLugano01] (robot YARP_DATA_DIRS)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubLugano01
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/robots/iCubLugano01] (robot YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/yarp/build-pc104/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/iCubContrib/share/ICUBcontrib/config/path.d] (robot path.d YARP_DATA_DIRS)
||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubLugano01/robotInterface.ini] (robot)
||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubLugano01/robotInterface.ini
"/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubLugano01/robotInterface.ini"

marijnfs, 2016-01-19:
Hi, what do I have to build to get the canBusAnalogSensor.ini file in the right place?

I also think these extra modules, like sharedCan, should be mentioned in the pc104-build wiki page, because now it seems incomplete.

marijnfs, 2016-01-19:
I rebuild all directories to the newest master branches (also because of the yarp bug) but now it still wont start. I added the output of robotInterface:

[output.txt](https://github.com/robotology/icub-support/files/95928/output.txt)

julijenv, 2016-01-28:
we can close it, nol?

closed 2016-01-28 by marijnfs