robotology/mesh-support #160: Occasional boot failure and canLoader error

mulkkyul, 2016-01-19:
Hi all,

We’re having some problems with our iCub (iCubDaejeon01). 
We replaced the tendons in the right thumb and now there are some problems when we try to calibrate the right thumb. We connected PC104 to the external monitor and observed the followings symptoms.

(1) Occasional Boot failure. 
 PC104 doesn’t start and does not proceed to linux-booting procedure. This problem occurs sporadically. When the booting fails, the current displayed on the power supply is 3.7~3.8 A. We ran badblocks and fsck on the USB but there was no error.

(2) PC104 hangs while executing S09_cfw2_sound.sh
Even though PC104 starts, it hangs during the boot procedure and does not show the login prompt. It hangs after showing “setting audio gain to 6”. We found that if we comment out the S09_cfw2_sound.sh in rciCub directory, PC104 doesn’t hang. The current was 2.3~2.5A when it starts normally.

(3) Problems in Calibration (Error connecting to cfw2can net2)?
After we commented out S09_cfw2_sound.sh, we were able to access PC104. Then, we executed canLoader and had problems connecting to cfw2can net2 (I think this is the right arm). We were able to connect to other nets, but when we tried to connect net 2, PC104 freezes. 

What would be the possible cause of those problems? and how can we check/fix? 
Also what/where should I look for the recent manual for calibration? (Wiki page seems a bit outdated)

Our power supply setting is:
PS1: preset voltage = 13V, preset power = 1.5kW, preset current = 15.5A, preset UVL = 10.66V, preset OVP = 15.4V
PS2: preset voltage = 39.2V, preset power = 1.5kW, preset current = 14.63A, preset UVL = 26V, preset OVP = 48V

Cheers
Jungsik

julijenv, 2016-01-28:
Hi @mulkkyul ,
Sorry for the late reply i was expecting someone more pc104 boot expert will give you some hints on the 2 first questions - I have no idea what could be the problems but what it is sure is that has nothing to do with the power settings, if it was working before it should normally make today with the same configuration.
Anyway, i'm gonna try to forward your mail to some pc104 analyst and see if they can understand of what u r experiencing ( @barbalberto @drdanz please guys try to help out to solve the 2 firsts questions i dont know if it is a problem with files inside the usb key or anything else , try to give some hints to solve it with your magic, thx)
Now, let's talk to your 3rd question, calibration procedure does not have anything to do with your issue so far from what i understand. You have something weird happening on your net2 if you cant connect to it. most probably broken wires, worst case scenario some boards burnt.
Here is the way to understand where is located the problem:
First try to detect a misbehavior when you turn on the motors switch and take a good care of the pace of the leds (you can see either by opening the back cover or by taking a look from the side of the robot).
Let me now your findings about this.
If nothing strange from right compare to left side.
try to test the can line in between the back of the head (N2 wire) and the 2 boards in the back of the robot (right side). Do you have continuity?
Could you check the state of the boards 2B1 and 2B0 (2 on the back right side of icub) as well as the 2B2 and the force torque sensor (you'll find those last board and sensor insde the upper arm right cover).
You could also try to seee once open the upper arm cover if the boards behave similarly (all of same if if 2 fo the stack of 3 boards you could find inside the cover are on different subnet).
Please make some pictures to help you out.
I am always behind my computer on skype iwth ID: icub.support if you want a fast reply.

Let us know

mulkkyul, 2016-01-28:
Hi @julijenv 
Thank you for the comments! 
I’ll check those things you mentioned and come back with some further details.

cheers,
jungsik

lornat75, 2016-02-03:
Hi @mulkkyul from the description of the problem 2) I guess you have a cable to connect the pc104 to a VGA screen, is this correct? What do you see on the screen when the boot fails (the problem 1) you describe). 

I think we can leave problem 2) for later 1) and 3) seems to be more urgent.

mulkkyul, 2016-02-04:
Hi @lornat75 
Yes, I connected the PC104 to a VGA screen and keyboard.

When the boot fails (Problem 1), there's nothing on the screen.
It seems the monitor doesn't receive signal from the PC104.

When the boot hangs while executing the shell scripts (Problem 2), this is the message on the screen.
[Picture](https://goo.gl/photos/syjxA1v3v7T8231G6)

Thanks,
jungsik

mulkkyul, 2016-02-15:
Hi @julijenv and @lornat75 
There are some updates on this issue.

**Regarding the (Problem 1. Occasional boot failure)**
We still have this problem and I found that there are 2 cases when the boot fails: 
- A monitor doesn't receive any signals from the PC104,
- A monitor receives signals but the PC104 hangs with showing a blinking cursor on a black screen.
Currently, the only solution we found is just wait 1 or 2 minutes. Then, sometimes it boots successfully. 

**Regarding the (Problem 3. Connecting cfw2can)**
As @julijenv suggested, we tested the several N2 wires behind the robot.
We found one disconnected wire (it was one of the wires in the right upper arm) and fixed it.

Now, I have another problem that connection to the can devices is unstable.
At canLoader, sometimes I am able to connect to the can devices including N2.
However, PC104 occasionally freezes when I click the connect button (regardless of the CAN number).

What I found is that when canLoader freezes, there's no message showing 
_"yarp: created device <cfw2can>. See C++ class Cfw2Can for documentation"_
Whenever it outputs that message on the screen, PC104 doesn't freeze and I can connect to the CAN devices.
I uploaded several pictures and videos of the screen and iCub on Google Photo:
[https://goo.gl/photos/syjxA1v3v7T8231G6]

What would be the possible cause of the problem? 
Thanks for your support.

julijenv, 2016-02-17:
Hi @mulkkyul ,
I reread from the first post of yours and something i didnt really much pay attentiion at first just just pop up in front of my eyes this time, I have a question:
The pc104 was working properly before you recabled the thumb proximal joint?
If yes, I assume that when you remounted the right arm you may have broke some wire that make this eratic behaviour. To solve this you may have to chack really and deeply the connection inside the upper right arm and looking for some broken wires.
To assert this you may have
1) First of everything , stop the pc104 and close the power motor switch.
2) Try to unconnect solely the net2 from the cfw2can board on the back of the head
3) remove the skin part and cover of the upper right arm (check the wires)
4) see if you can relaunch properly the pc104 without having to hang in the boot. (motor switch still OFF)
5) turn on the switch for the motors
6) repaet the canLoader test and tell us if you still succeed to freeze the pc104
if yes, close the motor switch and unconnect can0, can 1 (can2 already disconnected) can6 and can 8 connector from the cfw2 and redo step 1 to 6.
Let us know.

mulkkyul, 2016-07-04:
Hi @julijenv 

Sorry for really late feedback.. 

To answer your questions:
- PC104 was unstable even before the cable was broken. Somehow we solved the occasional boot problems by changing the PS settings. (Please see https://github.com/robotology/icub-support/issues/264 for our current PS settings.) In most cases, we can successfully start the PC104 board, but we still experience boot problems sometime.

- We tested canLoader as you suggested and we found that PC104 freeze unexpectedly. Usually, PC104 freeze when I tried to connect to any net. But sometimes, PC104 didn't freeze while connecting and it showed an error message saying "board not found". (This must be okay because I removed the all N# cables at the back of the head). I guess specific Net didn't cause the problem because connecting to same net number was sometimes successful and sometimes not. 
FYI, this is the log while I was testing canLoader.

*N2 cable at the back of the head was removed.

1. PC104 freezes while connecting to Net 0
2. PC104 freezes while connecting to Net 0.
3. PC104 didn't freeze while connecting to Net 3. But it freezes while connecting Net 1.
4. PC104 freezes while connecting to Net 1.

*N0, N1, N6, N8 cables were removed.

5. PC104 didn't freeze while connecting to Net 0, 1, 2 and 6. But it freezes while connecting Net 8.
6. PC104 freezes while connecting to Net 8.
7. PC104 freezes while connecting to Net 0
8. PC104 didn't freeze while connecting to Net 0,1,2,6,8. It freezes while connecting to Net 9. 
9. Boot failure. There was black screen showing blinking cursor. 
10. PC104 freezes while connecting to Net 0
11. PC104 freezes while connecting to Net 0.
12. PC104 didn't freeze while connecting to Net 0. But it freezes while connecting to Net 1.
13. PC104 didn't freeze while connecting to Net 0, 1, 8. But it freezes while connecting to Net 2.

*N5 and N7 cables are removed from the back of the head.

14. PC104 freezes while connecting to Net 0

*N3, N4, N9 cables are removed from the back of the head.

15. PC104 freezes while connecting to Net 0
16. PC104 freezes while connecting to Net 0

This is the screen when PC 104 didn't freeze while connecting.
[image]

mulkkyul, 2016-07-06:
Some updates on this issue:

- _dmesg_ shows different logs. Also our log seems different from the others (https://sourceforge.net/p/robotcub/mailman/message/29654269/)

**When PC104 is NOT freezing while connecting via canLoader**
[   77.936611] Cfw002 device driver V 0.6.3
[   77.936669] cfw002 0000:03:04.0: setting latency timer to 64
[   77.936705] Cfw002 found!
[   77.936709] Cfw002 registered with major 280, minor 0
[   78.011986] aerfx2_init
[   78.011989] sizeof(aerfx2_data) == 8
[   78.011990] INURB size: 512
[   78.011992] registering aerfx2 usb driver
[   78.012027] usbcore: registered new interface driver aerfx2
[   78.012029] aerfx2_init returns: 0
**[  181.296026] 0000:03:04.0 (cfw002): Firmware cfw002_fw.bin needed
[  181.306920] cfw002 0000:03:04.0: firmware: agent loaded cfw002_fw.bin into memory
[  181.306927] 0000:03:04.0 (cfw002): uploading 30599 bytes firmware
[  181.309724] 0000:03:04.0 (cfw002): image transfer done
[  181.932031] 0000:03:04.0 (cfw002): Got firmware alive rensponse after 156 ticks
[  181.938112] 0000:03:04.0 (cfw002) opened
[  185.364063] 0000:03:04.0 (cfw002) closed**

**When PC104 is FREEZING while connecting via canLoader**
[   80.569887] Cfw002 device driver V 0.6.3
[   80.571647] cfw002 0000:03:04.0: setting latency timer to 64
[   80.571713] Cfw002 found!
[   80.573357] Cfw002 registered with major 280, minor 0
[   80.655875] aerfx2_init
[   80.657115] sizeof(aerfx2_data) == 8
[   80.658312] INURB size: 512
[   80.659500] registering aerfx2 usb driver
[   80.660766] usbcore: registered new interface driver aerfx2
[   80.661678] aerfx2_init returns: 0

- This is the output of _lsmod_.

Module                  Size  Used by
[ ...... ]
cfw002                 21281  0 
[ ...... ]

- Testing the speaker volume freezes the PC104 board. 

As I mentioned earlier in this issue, I commented out "S09_cfw2_sound.sh". After boot, if I execute test_audio, PC104 freezes. (/lib/modules/3.2.0-4-686-pae/iCubDrivers/cfw002/LinuxDriver/tests/test_audio).

julijenv, 2017-08-14:
Hi @mulkkyul ,
Could you try to diconnect every can line of the cfw2 in order to uderstand if one of your can line is not short circuited; this is my fear.
After having disconnected all the CAN line just boot the pc104 itself with motors switch turn OFF (to prevent current flooding over without knowing).
on a shell of your server just ping pc104 until getting reply from it (you may have to wait a long time at the first restart)
1-If the pc104 starts, enter in it, ssh -X pc104, just wait a bit(some miuntes to check that the pc104 still responds) and write in a shell : sudo poweroff (type the pwd) - wait 1minute and switch OFF the pc104 button.
Redo the procedure 1 for 2 or 3 times to make sure that pc104 is working ok when the cfw2 is not in play.
2- now will start the debugging: please insert one can line at a time (from 0 to 1) and test one can line at a time (turn the motors switchON) and check if you can get canLoader to retrieve all the boards on the specific net without having the pc104 to crash or freeze. If one can line itself make the pc104 freeze, it means that you have some power cable disconnected that touch the specific can lines. you will have to investigate.
Let me know the result of your research.
Dont forget to always turn OOF the motors switch before unconnected the can line form the head.

maggia80, 2019-10-02:
let me close it. Fill free to reopen it in case needed

closed 2019-10-02 by maggia80