robotology/mesh-support #161: Hardware fault on robot startup

traversaro, 2016-01-19:
After updating everything on the `iCubGenova03` robot (`yarp`,`icub-main`,`icub-firmware-build`,`icub-firmware-shared` and after flashing the firmware with `updateRobot.sh`) the following all the torso, the legs joints are in hardware fault before calibration. Also the first 4 joints of the arms are in the same situation. 

The head calibrates correctly, while the wrist calibration is causing some violent movements of the wrist. 

This is the complete output of the `robotInterface`: 
~~~
693		57.884441				||| configuring
694		57.885600				||| no policy found
696		57.885720				||| default config file specified as robotInterface.ini
697		57.885741				||| checking [/home/icub/robotInterface.ini] (pwd)
698		57.885758				||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/robotInterface.ini] (robot)
699		57.885794				||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/robotInterface.ini
700		57.885817				||| finding file [config]
701		57.885837				||| checking [/home/icub/./icub_all.xml] (pwd)
703		57.885881				||| checking [/usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./icub_all.xml] (robot)
704		57.885904				||| found /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./icub_all.xml
706		57.885950				yarp: Port /log/pc104/robotInterface/4018 active at tcp://10.0.0.2:10008
708		57.886012				yarp: Sending output from /log/pc104/robotInterface/4018 to /yarplogger using tcp
709		57.886052		DEBUG		Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./icub_all.xml 
711		57.886129		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./icub_all.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
712		57.886148		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./icub_all.xml at line 2 . "robot" element should contain the "build" attribute [unsigned int]. Assuming 0 
713		57.886181		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./icub_all.xml at line 2 . "robot" element should contain the "portprefix" attribute. Using "name" attribute 
714		57.886204		DEBUG		Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml 
716		57.886250		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
717		57.886274		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
718		57.886297		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
719		57.886319		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/left_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
720		57.886343		DEBUG		Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml 
721		57.886379		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
723		57.886447		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . "devices" element should contain the "robot" attribute 
725		57.886510		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . Trying to import a file for the wrong robot. Found  instead of icub 
727		57.886537		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./cartesian/right_arm_cartesian.xml at line 1 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
729		57.886581		DEBUG		Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml 
731		57.886611		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
733		57.886643		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
735		57.886677		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
737		57.894384		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
739		57.894468		DEBUG		Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml 
741		57.894510		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
743		57.894557		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
745		57.894591		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
747		57.894625		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/right_arm_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
749		57.894660		DEBUG		Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_leg_mc_wrapper.xml 
751		57.894692		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_leg_mc_wrapper.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
753		57.894725		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_leg_mc_wrapper.xml at line 2 . "devices" element should contain the "robot" attribute 
755		57.894773		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_leg_mc_wrapper.xml at line 2 . Trying to import a file for the wrong robot. Found  instead of icub 
757		57.894819		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/left_leg_mc_wrapper.xml at line 2 . "devices" element should contain the "build" attribute [unsigned int]. Assuming 0 
759		57.894855		DEBUG		Reading file /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/right_leg_mc_wrapper.xml 
761		57.894897		WARNING		Invalid syntax while loading /usr/local/src/robot/icub-main/build-pc104/share/iCub/robots/iCubGenova03/./wrappers/motorControl/right_leg_mc_wrapper.xml at line 1 . No DTD found. Assuming version robotInterfaceV1.0 
[... cut ...]
3889		96.872226		DEBUG		sharedcan [6] thread ran 596 times, req.dT:10[ms], av.dT:10.06[ms] av.loopT :0.02[ms]
3891		96.872313		DEBUG		     Device has no ICanBusErr interface
3893		96.872367		DEBUG		sharedcan [6] printing boards infos:
3895		96.872417				None
3897		96.892420		DEBUG		Left_Arm_Calibrator: checkGoneToZeroThreshold: joint: 6 curr: 7350.813 des: 0.000 -> delta: 7350.813 threshold: 90.000 output: 0.000 mode: pos
3898		97.194463		DEBUG		sharedcan [3] thread ran 596 times, req.dT:10[ms], av.dT:10.06[ms] av.loopT :0.02[ms]
3900		97.194551		DEBUG		     Device has no ICanBu

traversaro, 2016-01-19:
A board reboot solved the hardware fault issue, but joints 4-7 of the arms still have the abovementioned calibration issues, and start in idle.

reafrancesco, 2016-01-21:
We had a look and it seems that updateFirmware.txt had wrong firmware version for many boards included the right and left arms. So if you @traversaro read that file to work out the firmware versions than this explains the problem you experienced. I have proceeded to flash the correct version of the firmware for the mentioned boards and THIS SOLVED THE PROBLEM

traversaro, 2016-01-21:
We should update the `firmwareUpdate.txt` file : 
https://github.com/robotology/icub-main/blob/master/app/robots/iCubGenova03/firmwareUpdate.txt to avoid further problems in the future.

traversaro, 2016-01-27:
This should have been fixed by https://github.com/robotology/icub-main/commit/6e1de9b7c2a395be4113d499e8eebb6d1d670a85 , please reopen if needed.
cc @reafrancesco

closed 2016-01-27 by traversaro