robotology/mesh-support #163: SFM & Eye fine-tunning

jypuigbo, 2016-01-19:
Hi, 

I'm trying to calibrate properly SFM but it almost never converges. Trying to see how that was happening, we realized that right camera is slightly looking upwards and to the right. This is specially noticeable when watching the two cameras' output, where one is tilted and with a vertical offset (where I understand the normal would be just horizontal offset for the eyes' separation). 

Is there any way to correct these offsets through calibration? Is there any way to correct this by hand?

Thanks, 

Jordi

pattacini, 2016-01-20:
@super-ste I guess there's a guide on that, right?
At least I remember I've seen it circulating some time ago ...

jypuigbo, 2016-01-22:
We specifically tried the [fine calibration of the head](http://wiki.icub.org/wiki/HeadFineCalibration) (but this can't solve the problem as it can't move one eye independent from the other) and the [stereo calibration of the eyes](http://wiki.icub.org/wiki/Stereo_calibration) which never gives a reasonable error. 

I attach a photo to show the problem with the eyes: 
[image]

pattacini, 2016-01-22:
Hi @jypuigbo 
From the above image I'd say that during calibration you could try:

1. Pan eyes towards left (robot side) up to the point when they both look diverging but of the same offset angle.
2. To compensate for this offset, start verging (positive values) until both eyes gaze reasonably straight.

super-ste, 2016-01-25:
@julijenv can provide additional method to mechanically compensate for pitch compensation between the two eyes if the advice from @pattacini does not solve.

jypuigbo, 2016-01-26:
@pattacini Looks a bit nicer, but it doesn't seem to improve performance... this is the ourcome: 
[image]

And for SFM, it always reports a failure on calibration, but seems normal to me, with the current eye positions. 
[image]

julijenv, 2016-01-26:
Hi @jypuigbo ,
remove the face cover and try to detect if the 2 screws for left eye or right eys got loose ( the ones from the picture below)?
If not try to loose the left or the right in order to improve the linearity in between the 2 eyes images (getting some degrees)., 
[image]

pattacini, 2016-01-26:
`SFM` will never work with those pairs of images, since they look **very very different**.

First, we have definitely to sort out the mechanical alignment issue.

jypuigbo, 2016-01-29:
Updates: 
Playing with those screws I managed to get a better camera calibration: 
[image]

The RMS error of the stereoCalib module was around 0.1, way better than before. 
Still the ```SFM``` is not able to calibrate properly and fails after 5 trials. 
Here's an example of the calibration view: 
[image]

@pattacini Do you think is good enough to save it? Do you need any extra information?

Still there is a small issue with the eyes that might be affecting. One of the cameras is slightly out of the eyeball (photos below). This makes that, for objects very near, there's a difference in size between both cameras. Any idea of whether this would be affecting the calibration and/or how to fix it?

[image]

pattacini, 2016-01-29:
Before struggling with `SFM`, I'd say we still need to clean up those images: to me they appear quite blurred. You could consider playing with the focus of the cameras, which - actually - will entail that you have to turn exactly those black parts in the bulbs clockwise or anti-clockwise.

@super-ste @julijenv other suggestions on this?

julijenv, 2016-01-29:
First thing first, You may need to focus the best as you can the image you receive. So as was refering @pattacini just tune the image by turning CLKwise or COUNTER-CLKwise until reaching the best quality image. I'dsuggest you to start with the eyes on which the lens is getting out the most from the bulb.
Let us knnow

jypuigbo, 2016-01-29:
@julijenv Does this imply I have to remove the whole eyeball? 
None of the cameras rotate on any of the directions, unless I unlock them, am I right?

julijenv, 2016-01-29:
Hi Jody,
you will have to first remove the iris of the eyebulb by uncrewing it, and you ll see afterwards that there is a screw above the lens holder to block the lens in position. untie it a little and you will be good to go for setting more properly the focus of the camera

clement-moulin-frier, 2016-02-08:
Hi guys,

We have tuned the focus of the eyes and relaunched the SFM calibration. This time we don't have any error when the calibration is running (it was failing before) but we are still not sure if the disparity map is good enough. Here is how it looks like:

[image]

It would be nice to have your input before going further with the tests. 
Thanks a lot.

pattacini, 2016-02-08:
Hi @jypuigbo and @clement-moulin-frier 

The disparity map doesn't look that bad, though it could be improved, I believe, by calibrating `SFM` with a scene containing objects of different sizes, filling also the height (i.e. _z-axis_). Actually, those objects ain't that big, hence the scene approximates a plane (that's bad for `SFM`).

Once calibrated, don't forget to save the results with the proper `SFM` rpc command.

pattacini, 2016-02-08:
You could check the 3D point associated to a 2D point in the left image doing the following:

- `yarpview --name /test --out /test/out`
- `yarp connect /icub/camcalib/left/out /test`
- `yarp read ... /test/out`
- click in the viewer `/test` on an object and note down the pixel `x y` read on the port
- `echo "x y" | yarp rpc /SFM/rpc`
- visually inspect whether the returned 3D point is reasonable.

Tariq-Abuhashim, 2016-02-08:
Also, seems like the images are still distorted (barrel distortion).
When calibrating the images, make sure the chessboard is covering as many pixels as possible, specially edges.

jypuigbo, 2016-02-16:
We recalibrated Cameras, SFM and the result looked like the following: 
```
[INFO] Running Left Camera Calibration... 
[INFO]RMS error reported by calibrateCamera: 0.22863
[INFO] Running Right Camera Calibration... 
[INFO]RMS error reported by calibrateCamera: 0.654199
[INFO]30 pairs have been successfully detected.
[INFO]Running stereo calibration ...
[INFO]done with RMS error= 2.630812
[INFO]average reprojection err = 3.834915
```

The calibration looked nicer, so we visually inspected different points as @pattacini proposed, with an average error of +/-2cm on any axis and a consistent offset on the Z axis of near -10 cm. Is this offset normal? There's any way of correcting it?

pattacini, 2016-02-16:
The `RMS` error of the right camera should be comparable to the one of the left ...
From which point on the robot you're measuring the z displacement?

clement-moulin-frier, 2016-02-18:
Hi,

Regarding the different `RMS` error for both eyes I don't know where it could come from. Maybe we haven't properly tuned the focus for the right eye, we're going to check this. Please let us know if you have another idea.

Regarding the Z displacement, we are measuring it from the point indicated by the `ROOT` node in [this picture](http://eris.liralab.it/images/9/97/NewiCubRefFrames1.png).

pattacini, 2016-02-18:
The `ROOT` node seems correct... thus, the error in the z-displacement should be in the same range of the other axes.

Try to calibrate the right camera again, in the meantime and then repeat the above experiment to see whether a better calibration will have any impact.

pattacini, 2016-02-22:
Hi guys, one important update on this.

When you rely on `SFM` to extract 3D points, you must make sure that the current vergence of the eyes is exactly the same as the one used while calibrating `SFM`.

This _magic_ value gets written within the `SFM.ini` as result of the calibration.

Hence, please find out your current `SFM.ini` (e.g. `yarp resource --context SFM --find SFM.ini`) file and check it out against this value.

pattacini, 2016-02-22:
##### Final note

To tell `ARE` to always keep vergence constant do:

**`ARE --motor::block_eyes 5.0`**

pattacini, 2016-03-07:
@clement-moulin-frier @jypuigbo @gregoire-pointeau 

We realized that offsets on z-axis are larger compared to those on the remaining axes. In fact, we also have something like `6-7 cm` of error on z against `2 cm` on x/y.

In end all these errors will be compensated resorting to `depth2kin` or "something equivalent".

I think we can then close this issue.

clement-moulin-frier, 2016-03-07:
Thanks for the update @pattacini 

We still have the issue of having different errors for each eye. I close the issue for now, we'll reopen it f needed.

closed 2016-03-07 by clement-moulin-frier

gsaponaro, 2016-03-23:
Thanks @pattacini, we'll give it a go next week. Just to confirm: for the fine calibration of the chains in question, we will refer to [this section of the manual](http://wiki.icub.org/wiki/Manual#Three._Calibration), while for ```SFM``` I guess it's about running the ```stereoCalib``` procedure with the checkerboard to get new intrinsic parameters. Also, do you use 320 or 640 resolution?

pattacini, 2016-03-23:
@gsaponaro 
you don't need to tune the intrinsic parameters again, hence I'd suggest you to rely on the `calibrate`+`save` rpc commands of `SFM`.