robotology/mesh-support #165: iCub firmware problem starting robotInterface

marijnfs, 2016-01-20:
After recompiling robotInterface it could not connect to the firmware, so I updated this. After seemingly successfully updating the firmware, it tells me it has problems finding the firmware version. I attached the output of robotInterface.

[output.txt](https://github.com/robotology/icub-support/files/97431/output.txt)

I get these messages which I also got before: 
*AxisName not found*
In another thread it said to recompile the latest YARP, which I did, but it didn't help.

But the main problem is with the firmware it seems, this is after running a firmware update on the robot:
*It has been detected that your control boards are not using the same
CAN protocol used by iCubinterface. iCubInterface cannot continue.
Please update your system (iCubInterface and/or your control board firmware.
For further information please visit: http://wiki.icub.org/wiki/Firmware*

Any suggestions for next steps?

julijenv, 2016-01-20:
Hi @marijnfs ,
actually the problem is as followed:
[[01;34mINFO[00m]iCubInterface CAN protocol: 1.4
[[01;34mINFO[00m]Firmware report:
[[01;33mWARNING[00m]sharedcan [6] joint: 0 can_address:  5 Unable to detect firmware version. Old firmware running?
[[01;33mWARNING[00m]sharedcan [6] joint: 1 can_address:  5 Unable to detect firmware version. Old firmware running?
[[01;33mWARNING[00m]sharedcan [6] joint: 2 can_address:  6 Unable to detect firmware version. Old firmware running?
[[01;33mWARNING[00m]sharedcan [6] joint: 3 can_address:  6 Unable to detect firmware version. Old firmware running?
[[01;34mINFO[00m]sharedcan [6] joint: 4 can_address:  7 board type: 3 (4DC) version: 1.30 build: 85 CAN_protocol:1.4
[[01;34mINFO[00m]sharedcan [6] joint: 5 can_address:  7 board type: 3 (4DC) version: 1.30 build: 85 CAN_protocol:1.4
[[01;34mINFO[00m]sharedcan [6] joint: 6 can_address:  8 board type: 3 (4DC) version: 1.30 build: 85 CAN_protocol:1.4
[[01;34mINFO[00m]sharedcan [6] joint: 7 can_address:  8 board type: 3 (4DC) version: 1.30 build: 85 CAN_protocol:1.4

The problem mainly is that one of the 2 MC4 (2B3) below the upper right arm cover csnt connect to the CAN Line.
Could you do a little check in between can5 and can6 (with canLoader software) to see that one board is missing, those 2 can Line should be symetrical in ID board version,release and build version?
To proceed easy peasy, jsut power cycle the motors switch and wait 10s for the closing and the reopening.
type canLoader in pc104 shell, select net5 and cfw2 and click connect (this is the list of boards connected to net5 this should be theoritically the same boards that you could also found on net6P)
Do a comparaison.
Let us know

marijnfs, 2016-01-20:
I compared after restarting the motors (just turn them of and on right?), and they are indeed not the same
[image]
[image]

julijenv, 2016-01-20:
Hi @marijnfs , 
As expected one of the 2 MC4 got missing on the right arm- U will have to open up the upper arm cover and check inside to see where the can Line got discontinued.
take a look at the board 2B3 and on the top part you have 2 connectors one 4ways and the other 8ways, check if  some wires got broken, pinched or anything. If you have difficulties you can send us picture and will try to help you out. You may have to remove the glue to make sure the continuity is alrgiht.
Let us know

marijnfs, 2016-01-21:
I don't see something clearly wrong on the cables. Is there a way to ignore the boards and boot up the robot anyway?

[image]
[image]

julijenv, 2016-01-21:
Hi @marijnfs ,
Could you perform a continuity test in between 4 ways pin connectors from board 2B3 and 2B4 and also the ssame test on 8 ways pin connector (raindow cables) , this should clear up what's going wrong.
Let us know

marijnfs, 2016-01-21:
How can I check this?

julijenv, 2016-01-21:
with a tester in continuity test mode with symbol noise

marijnfs, 2016-01-21:
Is the other board on the back?

marijnfs, 2016-01-21:
I tried finding the other end of the cable and opened the back, but it seems unreachable? What is the proper way to do this?

julijenv, 2016-01-21:
Please come on skype for a rapid call in roder to explain the procedure:
icub.support is our skype ID

marijnfs, 2016-01-21:
So the right board is working again, but the startup still complains about firmware:
[output.txt](https://github.com/robotology/icub-support/files/99172/output.txt)

marijnfs, 2016-01-21:
[updateRobot.txt](https://github.com/robotology/icub-support/files/99186/updateRobot.txt)

marijnfs, 2016-01-21:
// head V1
cfw2can 0 1 4DC.1.11.out.S
cfw2can 0 3 4DC.1.15.out.S

// torso
cfw2can 0 5 2BLL.1.54.out.S
cfw2can 0 6 2BLL.1.52.out.S

// left_arm
cfw2can 1 1 2BLL.1.50.out.S
cfw2can 1 2 2BLL.1.57.out.S
cfw2can 1 3 4DC.1.19.out.S

// left_hand
cfw2can 5 5 4DC.1.28.out.S
cfw2can 5 7 4DC.1.30.out.S

// right_arm
cfw2can 2 1 2BLL.1.50.out.S
cfw2can 2 2 2BLL.1.57.out.S
cfw2can 2 3 4DC.1.19.out.S

// right_hand
cfw2can 6 5 4DC.1.28.out.S
cfw2can 6 7 4DC.1.30.out.S

marijnfs, 2016-01-21:
Here is the output:
[output.txt](https://github.com/robotology/icub-support/files/99206/output.txt)

marijnfs, 2016-01-21:
Ok things seem to be moving so far! I still have a weird thing at startup where the head falls to the left (also happened before these problems). I can show you on skype?

julijenv, 2016-01-28:
Hi @marijnfs ,
it is the way the head V1 calibrate unfortunately that the reason why we move to V2 head.

closed 2016-01-28 by julijenv