robotology/mesh-support #166: Connector of Cable 5A1 broke

rueckert, 2016-01-20:
Hi,

the wires of the cable 5A1 in the left leg got loose. I attached a picture. Could someone please tell me how the wiring should look like. I can of course use the other leg as reference, however, I would like to get your experts opinion. 

As we do not have neither the right tools nor the connectors, I would try to solder the wires and connect them with the remaining metal pins in the connector. That is only a temporary solution, however, Morteza would like to get some data to continue with our collaborative work. 

Many thanks, Elmar
[image]

julijenv, 2016-01-20:
Hi @rueckert ,
Please add the label of your robot as a starter point like that we already know with who we are dealing with or if not available yet just write it down.
Second, please try to buy this list of tools and mainly the one in blue
[iCubMaintenanceTools (1).xlsx](https://github.com/robotology/icub-support/files/97618/iCubMaintenanceTools.1.xlsx)
now here the pinout of the cable which got snapped
[image]

good luck

rueckert, 2016-01-21:
Hi,

thank you very much for the swift and detailed instructions. I fixed the cable and all errors are gone.

closed 2016-01-21 by rueckert